ros-controls/ros2_control_demos #2: Organization -

olivier-stasse, 2020-04-26:
Dear @destogl ,
As mentioned during the ros2_control meeting I am willing to help.
From your designed document I tried to give it a shot in:
https://github.com/olivier-stasse/ros2_control_demos.git

It just a starting point to implement the generic headless robot taking inspiration from Dave Coleman and @Karsten1987 

Please feel free to reorganize or start from something different if you wish.

Let me know how you want to organize, we can probably split the work by issuing issue in the project https://github.com/orgs/ros-controls/projects/1

I will very much appreciate if you drive this but let me know if you are short in time.

Thanks again for your nice proposal.
Best,

bmagyar, 2020-04-26:
I'm happy to add you to the github project too @olivier-stasse ! @destogl could you please confirm if you have write permission to this repo or only to the github project?

olivier-stasse, 2020-04-27:
Dear @bmagyar if this fine with everyone this might be indeed interesting

destogl, 2020-05-05:
@bmagyar: I have write permission to this repo, thanks!

closed 2020-10-03 by bmagyar