ros-controls/ros2_control_demos #5: Create description for `demo_robot`

destogl, 2020-05-06:
* The descriptions follow the proposal from [roadmap design document](https://github.com/destogl/roadmap/blob/ros2_hardware_interface_design_proposal/design_drafts/hardware_interface.md)

Do the following:
* [ ] Define the `demo_robot` and describe it in short in README.md
* [ ] Create simple URDF to vizualize the `demo_robot`
* [ ] Create description for `ros2_control`-`Robot` class which uses "batch"-`headless` interface toward all its joints (i.e., `DemoRobotHeadless` class).
* [ ] Create description for `ros2_control`-`Robot` class which uses modular-`headless` interface toward its joints (i.e., `DemoSensorHeadless` and `DemoActuatorHeadless` classes).

destogl, 2020-06-02:
Check descriptions of
demo_robot.yaml
demo_robot_minimal.yaml
 in #21
@olivier-stasse : how does this look to you?

destogl, 2020-06-16:
@olivier-stasse here is something I would need help with, so if you have some time I would be very thankful :)

### Current state
* in #4 there is initial implementation of DemoRobot with URDF and xacro file
* I had problems with parsing xacro files in ROS2, therefore is comment about that in ros2_control_demo_robot/launch/minimal_robot.launch.py 
* I solved it by generating URDF in ROS1 and then using it directly

### What are next steps
- [ ] Use xacro file directly in ROS2
- [ ] Launch also robot_state_publisher
- [ ] Visualizes the `MinimalRobot` in rviz and check if the description make sense (test using joint_state_publisher is probably the easiest)
- [ ] Provide some fake funcitonality in `DemoRobotHardwareMinimal` class

When this works we should than concentrate on manager node and adding another examples.

Is this OK for you?

olivier-stasse, 2020-06-16:
Sounds good !
I will give it try.

Karsten1987, 2020-06-16:
I don't think it makes sense to parse xacro files. I think it makes sense to expect the `robot_description` to be presented as xml and therefore be parsed as such.

destogl, 2020-06-17:
Oh, I wrote that confusing...

I meant not to parse xacro file manually, but to convert it automatically in launch file into URDF file.

this didn‘t work for me with xacro in ROS2 for unknown reasons...

destogl, 2020-09-11:
Full list of examples: [source](https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md)

- [ ]    Industrial Robots with only one interface - #37 and ros-controls/ros2_control#140
- [ ]     Industrial Robots with multiple interfaces (can not be written at the same time) 2.1. Robots with multiple interfaces used at the same time - the same structure as in (2)
- [ ]    Industrial Robots with a sensor integrated into the robot's control box
- [ ]     Industrial Robots with a sensor connected to ROS computer
- [ ]    Modular Robots with separate communication to each actuator
- [ ]   Modular Robots with actuators not providing states and additional sensors (simple Transmissions)
- [ ]  Modular Robots with separate communication to each "actuator" with multi joints (Transmission Example) - (system component is used)
- [ ] Sensor only
- [ ]  Actuator Only