ros-controls/ros2_control_demos #16: Resolve double checks in CI, Linters and Coverage

destogl, 2020-05-07:
There is overlap in the tests when running CI, Linters and Coverage actions. It would be nice to resove this at some point.

ideas?

olivier-stasse, 2020-05-08:
Could you be a bit more specific ?

destogl, 2020-05-08:
If you look at the actions it seams that the Lint-Tests are called in every configuration. This is probably because of `colcon test` call. I am not sure if this should/can be removed so that we have one specific action broken and all of them when some of the Lints are not happy.

olivier-stasse, 2020-05-08:
It seems to come from 
https://github.com/ros-controls/ros2_control_demos/blob/fe783a869a8f5db0edbe7c954b19250b148f18a3/ros2_control_demo_robot/CMakeLists.txt#L26
for the ros2_control_demos package.
If it can be removed then it could set by a cmake option. Then the github action in charge of the linter would set it to TRUE.

bmagyar, 2020-05-10:
Indeed that line breaks the build_and_test step when a linter fails which shouldn't really be the case as it goes against civilized CI setups where build, test and linting are separate stages. For technical reasons build and test cannot be separated and unfortunately due to the strange habit of linting being shoved into tests, it's all bundled up. 

I however support removing linting from the tests section in the cmake, the CI is set up to run linting steps separately already.

destogl, 2020-05-10:
I will remove this line from CMakeLists.

destogl, 2020-05-10:
I removed `lint` dependencies compleately in #18 .

closed 2020-05-11 by bmagyar