ros-controls/ros2_control_demos #42: Demo not work

LuisYordano, 2020-10-01:
I'm testing demo -b add_rrbot_system_position_joints.

In the file rrbot_system_position_only.launch.py, exist line executable='ros2_control_node', but in the package control_manager **not exist ros2_control_node**.

[image]

Could indicate the steps to run the demo correctly.

Thank you in advance.

destogl, 2020-10-02:
Hi @LuisYordano,

thanks for your interest in the ros2_control.

The instructions are in the ros-controls/ros2_control#147  at the bottom of the [first post](https://github.com/ros-controls/ros2_control/pull/147#issue-487193575).

Please consider that the library is currently under heavy development, therefore the API could by broken at any time!

LuisYordano, 2020-10-02:
Hi @destogl,

Thanks for the support.

Error testing demo : 

`[ERROR] [1601656357.549508851] [ControllerManager]: Loader for controller 'forward_command_controller_position' not found
`

[image]

I have executed the code without making any modifications.

As the library is under development, I close this issue.

closed 2020-10-02 by LuisYordano

destogl, 2020-10-02:
Did you sourced your environment after building the packages?

LuisYordano, 2020-10-03:
Yes, I did.

Could indicate the links of the repositories that works for you?