ros-controls/ros2_control_demos #50: Create demo for diff drive controller

destogl, 2020-12-22:
__Originally posted by @bmagyar in https://github.com/ros-controls/ros2_controllers/pull/131__

BorgesJVT, 2021-02-04:
Is anyone in this? 
Could you provide more details? 
For example, could I create a "diffbot" folder next to rrbot? Using something like ros2_control_demo_hardware/DiffBotSystemHardware?

bmagyar, 2021-02-08:
The idea is to build a demo with a mobile base similar to rrbot. We used diffbot for this traditionally. I am not aware of anyone working on this ATM

destogl, 2021-02-08:
> Is anyone in this?

Would be great if you could jump on this.

> Could you provide more details?

- A hardware in ros2_control_demo_hardware should be created
- Launch file should be added into ros2_control_demo_robot

joshnewans, 2021-02-21:
Hi guys, I've been looking to chip in and struggled to find the right place (answers, discourse, email, etc.) so I thought I'd put it in here.

I had been working on writing a reasonably beginner-level tutorial on how to build a differential drive robot with ROS, starting with a URDF, placing that in Gazebo with virtual actuators and sensors, hooking it up to a diff_drive_controller, building a physical robot, then using the same controller to control the real robot (and do nav etc.). As part of this I created a package called [diffdrive_arduino](https://github.com/joshnewans/diffdrive_arduino/) which provides a suitable `hardware_interface` for an Arduino running the firmware from `ros_arduino_bridge`.

I had my platform mostly ready but some of the dependent packages weren't released for noetic, and I had some issues with melodic, so I decided to migrate the whole thing to ROS2. I was a bit confused before I realised that `ros2_control` isn't quite at as mature as the original version, but I see this as an opportunity to get my hands dirty and help out! (I hopped in on your meeting the other day to get a vibe for where things are at.)

I have ported my code to work on `ros2_control` and it seems to run fine on my test platform (barring a few minor issues I can raise), but as far as I can tell there is currently no way to do the Gazebo part of my tutorial, and I also found the whole plugin setup to be a bit confusing (compared to `ros_control` at least).

So I'd like to help out if I can, either by:
- Writing code (I'm reasonably experienced with C++ but fairly new to ROS and especially `rclcpp`).
- Testing the `diff_drive_controller` in Gazebo and on my demo bot
- Writing some official documentation alongside my tutorial

I'm fairly limited in doing this in my spare time, but I'm quite keen to get involved so please let me know what I can do. (And if a gazebo plugin exists, please point me in the right direction).

Thanks

destogl, 2021-02-22:
Hi @joshnewans, thanks for your interest in contributing to the ros2_control!

I see the following opportunities for you to help:

- review #70 
- Writing example URDFs for Diff drives into the [roadmap repo](https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md)
  - There are two options, first is describing the mobile base as a system (#70) see Examples 1-3 and, second, describing robot as a set of actuators and sensors (modular approach - Example 5)
- Implement examples for these architectures and document them
 
> I have ported my code to work on `ros2_control` and it seems to run fine on my test platform (barring a few minor issues I can raise), but as far as I can tell there is currently no way to do the [...]

The Gazebo currently supports SystemHardwareInterface. This is the first implementation and there are for sure possibilities to improve it. For details please check the gazebo_ros2_control repo (or similar name).
 
>     * Writing code (I'm reasonably experienced with C++ but fairly new to ROS and especially `rclcpp`).
The easiest way to start contributing is to tackle the issues marked as 'Good First Issue'.

>     * Testing the `diff_drive_controller` in Gazebo and on my demo bot
This would be great!

>     * Writing some official documentation alongside my tutorial
We have official docs repo at ros-control/control.ros.org. We are planning to port also the designs from the roadmap repo, when they are fixed.

> I'm fairly limited in doing this in my spare time, but I'm quite keen to get involved so please let me know what I can do. (And if a gazebo plugin exists, please point me in the right direction).
Please check the gazebo_ros2_control repo. and raise issues there for any questions regarding gazebo simulation.

joshnewans, 2021-02-27:
Thanks @destogl for your feedback!

PR #70 (2021-04-28): demo_diff_drive_controller
Resolves #50 

- A diffbot folder was created similar to rrbot for URDF files;
- A hardware in ros2_control_demo_hardware was created with position and velocity states and velocity command;
- Launch file was added into ros2_control_demo_robot.
closed 2021-04-28 by destogl