ros-controls/ros2_control_demos #55: Gazebo examples

C-monC, 2021-01-19:
Hi,

Are the gazebo files currently not on the repo? Searching for ros2_control_demo_hardware_gazebo returns nothing. 

Thanks,
Simon

destogl, 2021-01-19:
Hi Simon,

no they are not. The gazebo support is currently in [development](https://discourse.ros.org/t/gazebo-support-for-ros2-control-breakout-session/18468). We should have this in the next 4-6 weeks.

C-monC, 2021-01-19:
Okay cool, is tomorrow's meeting open? I am interested in helping.

closed 2021-01-19 by C-monC

destogl, 2021-01-19:
> Okay cool, is tomorrow's meeting open? I am interested in helping.

Of course! You are very welcome to join! Also check the work for @ahcorde in the ros-simulation/gazebo_ros2_control#44

C-monC, 2021-01-19:
Perfect, I'll definitely join. 
Are there some easier/smaller issues I can try? I haven't done much c++ apart from tutorials/courses.

destogl, 2021-01-19:
If you want to start, look are issues marked with "good first issue" label. They are usually easy and straightforward