ros-controls/ros2_control_demos #59: RRBotSystemMultiInterface example with multiple command interfaces

mahaarbo, 2021-02-01:
I thought I'd try to make a [RRBotSystemMultiInterface](https://github.com/mahaarbo/ros2_control_demos) according to the [components_architecture_and_urdf_examples](https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md) example that has position, velocity, and acceleration. Integration of the velocity or acceleration to position depends on whether these interfaces have been claimed, but how can the hardware interface see which command interfaces are currently claimed, or react to changes of the command interfaces claimed? 

In the example, the problem pops up here: [rrbot_system_multi_interface/L190-L200](https://github.com/mahaarbo/ros2_control_demos/blob/2eb6b99ca0de78c0a57cb528d0d8d4d1da82df2e/ros2_control_demo_hardware/src/rrbot_system_multi_interface.cpp#L190-L200). This problem also popped up in my attempts to port Franka Emika Panda to ROS2. Previously the RobotHW would be able to notice the switch of controller in `prepareSwitch` and `doSwitch`, how should it be done with the new resource manager standing between the hardware_interface and the controller manager?

olivier-stasse, 2021-02-02:
IHMO I would add a field in the multiple interface in order to switch from one to the other.
I tried an example here:
https://github.com/olivier-stasse/ros2_control_demos/blob/35a4bd6092434185653eefc8ed1a9a07e13bb846/ros2_control_demo_hardware/include/ros2_control_demo_hardware/rrbot_system_quadruped.hpp#L51

Currently I am not completely sure if it is possible to use another type than `double` to interact with the control interface.

From a control viewpoint you may need to have position and velocity for stabilization and/or good tracking performances in effort control mode.

olivier-stasse, 2021-02-02:
The main limitation to change the control interface type is coming from here:
https://github.com/ros-controls/ros2_control/blob/f506c2a2a449744ab4e9a414090abb7c182adee0/hardware_interface/include/hardware_interface/loaned_command_interface.hpp#L64

The impact of adding a template parameter to this class is unclear for me.

destogl, 2021-02-03:
Hi @mahaarbo, 

thank you for your interest in `ros2_control`!

IMO, it seems you found a missing feature. Since the switch is managed completely by the CM in the ros2_control there is a missing interface to notify resources about controllers' states. Should we have some kind of "notify_about_controller_switch" method?

Could you explain, what would be a suitable process for you to notify resources when switching controllers? We can then discuss this in the WG meeting next Wednesday and may already propose some solution to it.

destogl, 2021-02-03:
> The example do not do the switch but extend the interface.

@olivier-stasse This is also a solution. Would you prefer some kind of "claimed-command-interfaces" notification or to go with extending interfaces.

> Currently I am not completely sure if it is possible to use another type than `double` to interact with the control interface.
> Because you would probably want to use something else than a double for type enumeration over a possible set of control implemented in the power electronics.

The interfaces are currently limited to double type. Extension to bool and/or int type should be a no-brainer but strings would be probably not possible.

> From a control viewpoint you may need to have position and velocity for stabilization and/or good tracking performances in effort control mode. 

Should than controller claim all interfaces? Or is it sufficient to know this in the hardware? Probably there needs to be some kind of blocking mechanism also for other, non-claimed, interfaces.

@bmagyar, @Karsten1987 please take also a look at this and share your opinion.

mahaarbo, 2021-02-03:
@destogl and thank you for the great work you guys have put into `ros2_control`, it feels as though it will be a cleaner system than `ros_control` :)

As for how to implement this, I'm not entirely sure. One way would be to allow the user to specify a function in the `hardware_interface` which is associated with a `command_interface` and that will be called by the `resource_manager` on `claim_command_interface`, but I don't know if that will be complicate the system too much? As each joint has its own `command_interface`, the logic to handle the resources in a good way would again fall back to the user's implementation of the `hardware_interface`.

destogl, 2021-02-03:
@mahaarbo Do you see a need to have two functions, e.g., "check_claim_interfaces" and "clam_interfaces_notification" or the latter is sufficient for your scenario?

mahaarbo, 2021-02-03:
@destogl I can try to make an example draft for `ros2_control` if you'd like.

mahaarbo, 2021-02-04:
@destogl Here is a working example: [mahaarbo/ros2_control/notify_resource_claim](https://github.com/mahaarbo/ros2_control/tree/notify_resource_claim_example) with the demos from [mahaarbo/ros2_control_demos/notify_resource_claim](https://github.com/mahaarbo/ros2_control_demos/tree/notify_resource_claim_example). The resource manager will call a `notify_state_resource_claim` and `notify_command_resource_claim` when a resource is claimed. This in turn calls an `accept_state_resource_claim` method  and an `accept_command_resource_claim` in all relevant components. I'm a little unsure about doing it this way, since it becomes a required function that everyone has to implement, even if most cases may not need it (since they don't expose too many command interfaces, and likely don't have to change the communication with the hardware when different resources are claimed).

destogl, 2021-02-04:
@mahaarbo: This looks great! thanks! Just one question:

#. Is this relevant for the `state` interfaces? shouldn't they always be populated? Are there some complex HW-scenarios where many states have to calculate?

I propose to proceed as follows:
#. Can you rename/change functions to have the following signature?
```
return_type notify_state_interfaces_claim(const std::vector<std::string> & interfaces)
return_type notify_command_interfaces_claim(const std::vector<std::string> & interfaces)
```
#. Can you rebase your new branch to ros2_control's master?
#. Can you open PR to ros2_control, please? We can continue the discussion there over the code :)

> I'm a little unsure about doing it this way since it becomes a required function that everyone has to implement, even if most cases may not need it (since they don't expose too many command [...]

This could be solved using [`BaseInterface` class](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/base_interface.hpp). Although not implementing the functions with final so people can override it.

olivier-stasse, 2021-02-04:
@mahaarbo Thanks a lot for this, it really helps in clarifying the use-cases.
If I am not mistaken you allow the system to claim position, velocity and effort control interfaces independently, i.e.
it is not possible to acquire the position command interface of a joint by two controllers.
But it is possible to acquire the position command interface and the effort command interface of a joint by the same controller. They are not mutually exclusive.

olivier-stasse, 2021-02-04:
IMHO it would be preferable to avoid imposing the existence of classes such as the PosVelAccJointHandle class in ros1_control:
https://github.com/ros-controls/ros_control/blob/noetic-devel/hardware_interface/include/hardware_interface/posvelacc_command_interface.h

destogl, 2021-02-04:
@olivier-stasse I understood there is an issue that HW does not know which interfaces are currently claimed and therefore it doesn't know how to control the robot, i.e., which values should be used, e.g., position or velocity. 

The case you are describing, with not mutually exclusive interfaces, should be resolved on the controllers' level. So controller should ask all interfaces which are not mutually exclusive. Could this work like this? Or do you also see some other issues?

mahaarbo, 2021-02-04:
>  * Is this relevant for the `state` interfaces? shouldn't they always be populated? Are there some complex HW-scenarios where many states have to calculate?

I'm not entirely sure, I'll remove the state notifications for now, and then it can perhaps be included later if the need arises :)

I'll rebase and then we can continue the code discussion in the pull request.

> They are not mutually exclusive.

Yup! Then the same example can be used with two `forward_command_controller`s using different command interfaces on the two different joints to illustrate [robots with multiple interfaces used at the same time](https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md#21-robots-with-multiple-interfaces-used-at-the-same-time---the-same-structure-as-in-2). I think it is reasonable to assume that the `hardware_interface` creator will be the best one to judge which combination of command interfaces can be used together.

olivier-stasse, 2021-02-04:
> @olivier-stasse I understood there is an issue that HW does not know which interfaces are currently claimed and therefore it doesn't know how to control the robot, i.e., which values should be used, [...]

They are two very different pratical use cases:
- Either the PID is done at the controller level i.e. the CPU working at 1KHz (max 2.6 KHz) and then the solution you are describing is correct.
- Or, which appears quite frequently, the PID or a cascade of controllers is done at the power electronics level (using micro controller) at much higher frequency (10-20 KHz) and then this has to be at the hardware level. The hardware interface needs to interact with the bus of the power electronics which is often a CAN bus, a high-speed serial bus or an Ethercat.

We have both cases in my lab. The second case is appearing more frequently with the rise of very powerful microcontrollers. 

I may have miss an important part of our architecture in how to deal with the second case. In this case I will be delighted to have explanation on how to solve this.

olivier-stasse, 2021-02-04:
@destogl May be there is a need of a negociation phase between the controller and the hardware ?

Implictly the effort controller is asking the hardware to switch to an only effort command interface.

One simple way would be to check if the combination of claim command interface can be handle by the hardware when switching to a controller.

olivier-stasse, 2021-02-04:
> Yup! Then the same example can be used with two `forward_command_controller`s using different command interfaces on the two different joints to illustrate [robots with multiple interfaces used at the same time](https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md#21-robots-with-multiple-interfaces-used-at-the-same-time---the-same-structure-as-in-2). I think it is reasonable to assume that the `hardware_interface` creator will be the best one to judge which combination of command interfaces can be used together. Edit: I see now that I misunderstood, I read too quickly and thought it was about mutually exclusive regarding claiming position interface or velocity interface on all joints at once. I agree that certain controllers may need to claim multiple interfaces for a single joint.

Thanks a lot for the explanations.

Karsten1987, 2021-03-24:
sorry, I am a bit late to the party here.

I am somewhat sticking to my opinion that I see the actual hardware implementations for each actuator, sensor and system as simple as possible and mostly linear in its execution. That is, having a support for a callback in each hardware interface projects quite some business logic on each implementation to guarantee a real time safe execution and necessarily some multithreaded logic (most likely with custom locks & mutexes).
Even if you'd want to get rid of the threads and assume that `read`, `write` is never being called at the same time than the notification callback this implies a strong coupling to the resource manager and still doesn't make it linear in its execution.

Why can't we leverage the lifecycle methods for the hardware interfaces for this? When the CM/RM receive a switch command, it's its responsibility to stop existing hardware execution by a call to `stop`, assign the new resources and call resume aka `start` on the hardware interface. We could obviously introduce some finer granularity for the lifecycle to separate a `start` & `stop` from a `reconfigure`. But in any case that way it's still linear.
I mentioned here resources to reconfigure, but that also implies further configurations which are not strictly bound to resources but rather various parameters for the hardware.

Alternatively, I could see a polling mechanism in which each hardware gets a shared config options struct or something (shared with the RM) in its configure step. And instead of implementing a concurrent and async callback we rather have a polling function from the hardware which can check the state of that shared config to actively react on state changes.

mahaarbo, 2021-04-29:
Now that #83 has been merged, it's time to close this issue.

closed 2021-04-29 by mahaarbo