ros-controls/ros2_control_demos #81: RRBOT Position and Velocity Controller

gftrobots, 2021-04-21:
Dear all, I'm struggling to get the position only turned into position and velocity.
Can one of you advise how to expand "rrbot_system_position_only" to cover position and velocity control?

gftrobots, 2021-04-22:
This is an example description file:
`<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:macro name="rrbot_system_position_only" params="name prefix slowdown:=2.0">

    <ros2_control name="${name}" type="system">
      <hardware>
        <plugin>ros2_control_demo_hardware/RRBotSystemPositionVelocityHardware</plugin>
        <param name="example_param_hw_start_duration_sec">2.0</param>
        <param name="example_param_hw_stop_duration_sec">3.0</param>
        <param name="example_param_hw_slowdown">${slowdown}</param>
      </hardware>
      <joint name="joint1">
        <command_interface name="position">
          <param name="min">-1</param>
          <param name="max">1</param>
        </command_interface>
        <command_interface name=“velocity”>
          <param name="min">-1</param>
          <param name="max">1</param>
        </command_interface>
        <state_interface name="position"/>
        <state_interface name=“velocity”/>
      </joint>
      <joint name="joint2">
        <command_interface name="position">
          <param name="min">-1</param>
          <param name="max">1</param>
        </command_interface>
        <command_interface name=“velocity”>
          <param name="min">-1</param>
          <param name="max">1</param>
        </command_interface>
        <state_interface name="position"/>
        <state_interface name=“velocity”/>
      </joint>
    </ros2_control>

  </xacro:macro>

</robot>`

mahaarbo, 2021-05-05:
Since you have both a position command interface and a velocity command interface, perhaps you could look at the [`rrbot_system_multi_interface`](https://ros-controls.github.io/control.ros.org/ros2_control_demos/doc/index.html) demo? That is a hardware interface that supports mode switching (using either position or velocity commands).

destogl, 2023-02-02:
obsolete

closed 2023-02-02 by destogl