ros-controls/ros2_control_demos #88: Running this I get

FirstCaptain, 2021-04-28:
I am seeing the following errors and the rrbot is failing to load in RVIZ after following instructions in the Readme

My System is Ubuntu 20.04 and I am running Ros2 Foxy. 

[image]
[image]

Karsten1987, 2021-04-28:
I've encountered this as well and I feel we should update some documentation on this.

The system is not broken, it's just not fully activated yet. If you look into the launch file, we're launching a robot state publisher node. That node however needs a `joint_states` topic to be published on. For this to happen, we need to load and start the `joint_state_controller`, alternatively also called `joint_state_broadcaster`. 

All you have to do is to call the following in a second shell after you've launched your system:
```
ros2 control load_controller joint_state_controller --state start
```

FirstCaptain, 2021-04-29:
Awesome thanks, that works

bmagyar, 2021-04-30:
Let's keep this open until docs are updated

destogl, 2023-02-02:
obsolete.

closed 2023-02-02 by destogl