ros-controls/ros2_control_demos #91: gzserver segmentation fault

Cdfghglz, 2021-05-02:
I have followed the instructions and wanted to try the Gazebo example with

`ros2 launch ros2_control_demo_robot rrbot_system_position_only_gazebo.launch.py`

Versions as of issue creation time:
 * Kubuntu 20.04, Foxy apt packages up to date
 * _control, _controllers, _control_demos on a fresh master in colcon_ws
 * gazebo 11.3.0

The output:
```
[INFO] [launch]: All log files can be found below /home/cdf/.ros/log/2021-05-02-16-30-58-135425-xps15-15435
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [15437]
[INFO] [gzclient   -2]: process started with pid [15439]
[INFO] [robot_state_publisher-3]: process started with pid [15442]
[INFO] [spawn_entity.py-4]: process started with pid [15445]
[INFO] [spawner.py-5]: process started with pid [15447]
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_link had 1 children
[robot_state_publisher-3] Link link1 had 1 children
[robot_state_publisher-3] Link link2 had 3 children
[robot_state_publisher-3] Link camera_link had 1 children
[robot_state_publisher-3] Link camera_link_optical had 0 children
[robot_state_publisher-3] Link hokuyo_link had 0 children
[robot_state_publisher-3] Link tool_link had 0 children
[robot_state_publisher-3] [INFO] [1619965858.843488088] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1619965858.843573769] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-3] [INFO] [1619965858.843583449] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-3] [INFO] [1619965858.843589613] [robot_state_publisher]: got segment hokuyo_link
[robot_state_publisher-3] [INFO] [1619965858.843594700] [robot_state_publisher]: got segment link1
[robot_state_publisher-3] [INFO] [1619965858.843599554] [robot_state_publisher]: got segment link2
[robot_state_publisher-3] [INFO] [1619965858.843604367] [robot_state_publisher]: got segment tool_link
[robot_state_publisher-3] [INFO] [1619965858.843609273] [robot_state_publisher]: got segment world
[spawner.py-5] [INFO] [1619965859.050887463] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawn_entity.py-4] [INFO] [1619965859.288095445] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1619965859.288498049] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] [INFO] [1619965859.290043114] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1619965859.302540653] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1619965859.302919486] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1619965859.808193191] [spawn_entity]: Calling service /spawn_entity
[gzserver-1] [INFO] [1619965860.352477460] [camera_controller]: Publishing camera info to [/camera1/camera_info]
[spawn_entity.py-4] [INFO] [1619965860.358004668] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [rrbot_system_position]
[gzserver-1] [WARN] [1619965860.378583637] [gazebo_ros_head_hokuyo_controller]: missing <output_type>, defaults to sensor_msgs/PointCloud2
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 15445]
[gzserver-1] [INFO] [1619965860.483478340] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1619965860.486060309] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1619965860.486105794] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1619965860.486136422] [gazebo_ros2_control]: Loading parameter file /home/cdf/ws/colcon_ws/install/ros2_control_demo_robot/share/ros2_control_demo_robot/config/rrbot_gazebo_forward_controller_position.yaml
[gzserver-1] 
[gzserver-1] [INFO] [1619965860.488876630] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1619965860.489500242] [gazebo_ros2_control]: Recieved urdf from param server, parsing...
[gzserver-1] [INFO] [1619965860.508523322] [gazebo_ros2_control]: Loading joint: joint1
[gzserver-1] [INFO] [1619965860.508580270] [gazebo_ros2_control]:       Command:
[gzserver-1] [INFO] [1619965860.508591721] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1619965860.508600289] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1619965860.508607975] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1619965860.508617356] [gazebo_ros2_control]: Loading joint: joint2
[gzserver-1] [INFO] [1619965860.508625250] [gazebo_ros2_control]:       Command:
[gzserver-1] [INFO] [1619965860.508632212] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1619965860.508639787] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1619965860.508646536] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1619965860.512728768] [gazebo_ros2_control]: Loading controller_manager
[gzserver-1] [WARN] [1619965860.545596888] [gazebo_ros2_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-1] [INFO] [1619965860.545711655] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
[gzserver-1] Segmentation fault (core dumped)
[ERROR] [gzserver-1]: process has died [pid 15437, exit code 139, cmd 'gzserver                                                                       -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
```

Could anyone help?

Karsten1987, 2021-05-13:
Can you launch your system with a verbose output of gazebo:
Turning this line to `True`: https://github.com/ros-controls/ros2_control_demos/blob/master/ros2_control_demo_robot/launch/rrbot_system_position_only_gazebo.launch.py#L36

do you also have gazebo_ros2_control packages built from source?

Cdfghglz, 2021-05-15:
Hi, thanks for your inputs.

Here is some of the the output of a verbose gazebo run:

```
[gzclient --verbose  -2] [Msg] Waiting for master.
[gzclient --verbose  -2] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gzclient --verbose  -2] [Msg] Publicized address: 192.168.84.160
[gzclient --verbose  -2] [Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/cdf/ws/colcon_ctrl_ws/install/ros2_control_demo_robot/share/ament_index"
[gzclient --verbose  -2] [Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/cdf/ws/colcon_ctrl_ws/install/ros2_control_demo_robot/share/colcon-core"
[gzclient --verbose  -2] [Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/cdf/ws/colcon_ctrl_ws/install/ros2_control_demo_robot/share/ros2_control_demo_robot"
[gzserver-1] [WARN] [1621055372.290500959] [gazebo_ros2_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-1] [INFO] [1621055372.290712444] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
[gzclient --verbose  -2] [Err] [Connection.cc:546] Connection[5] Closed during Read
[gzclient --verbose  -2] [Err] [TransportIface.cc:385] Unable to read from master
[gzclient --verbose  -2] [Err] [GLWidget.cc:896] Unable to connect to a running Gazebo master.
[gzserver-1] Segmentation fault (core dumped)
[ERROR] [gzserver-1]: process has died [pid 92701, exit code 139, cmd 'gzserver       --verbose                                                                -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[gzclient --verbose  -2] [Wrn] [Publisher.cc:136] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.
```

Then I additionally cloned gazebo_ros2_control to the colcon workspace (previously I used system install), but got an error building it:
```
In file included from /usr/include/gazebo-11/gazebo/msgs/MessageTypes.hh:7,
                 from /usr/include/gazebo-11/gazebo/physics/Joint.hh:29,
                 from /home/cdf/ws/colcon_ctrl_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system_interface.hpp:23,
                 from /home/cdf/ws/colcon_ctrl_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:25,
                 from /home/cdf/ws/colcon_ctrl_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_system.cpp:19:
/usr/include/gazebo-11/gazebo/msgs/altimeter.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/altimeter.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers.  Please
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/altimeter.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
```
`protoc --version` returns `libprotoc 3.14.0`

Cdfghglz, 2021-05-16:
I managed to build gazebo_ros2_control by following step 1. in https://github.com/AcutronicRobotics/gym-gazebo2/issues/13#issuecomment-482410238

The gazebo segmentation fault still persists however, producing the same exact verbose output.

eliabruni, 2021-05-19:
Hello, having the same issue.

bmagyar, 2021-05-20:
Are you guys trying to compile things that clash on the protobuf compiler version? That's going to be a though nut to crack... I am not sure we can help you from ros2_control, essentially anything using gazebo will be broken. Are you trying to use gym + gazebo + ros2_control? That description seems to apply to ROS Crystal and Ubuntu 18, I'm not sure if it applies to Foxy and Ubuntu20 as well.

Cdfghglz, 2021-05-21:
The compilation errors were solved by following the link referenced in my earlier reply https://github.com/ros-controls/ros2_control_demos/issues/91#issuecomment-841851095

The problem happens on runtime. It possibly still has to do with protobuf. Over the weekend I try to set up everything clean from scratch with a single version of protobuf on my system and report.

olivier-stasse, 2023-03-26:
Closing the issue for lack of activity. Feel free to reopen it if needed.

closed 2023-03-26 by olivier-stasse

yoojinPanda, 2023-03-28:
Hi. I'm having the exactly same issue here... did anyone solve this problem?