ros-controls/ros2_control_demos #97: Issues with demos

nicholaspalomo, 2021-06-02:
Hi, everyone,

Here are some issues in the demos and the changes I made to get them to work:

- [ ] In step 2 under the heading "Using ForwardCommandController" of the ReadMe, 

```ros2 control switch_controllers --start-controllers forward_position_controller```
should be changed to:
```ros2 control switch_controllers --start forward_position_controller```

- [ ] In step 3 of the same section of the ReadMe:
```ros2 launch ros2_control_test_nodes rrbot_test_forward_position_controller.launch.py```
should be changed to:
```ros2 launch ros2_control_demo_robot test_forward_position_controller.launch.py```

- [ ] In `test_forward_position_controller.launch.py`, lines 25-29 should be changed to:
```
position_goals = os.path.join(
        get_package_share_directory("ros2_control_demo_robot"),
        "config",
        "rrbot_forward_position_publisher.yaml",
    )
```

- [ ] In `rrbot_controllers.yaml`, line 14 should be changed from:
```forward_position_controller```
to:
```/forward_position_controller```.

- [ ] In `rrbot_system_position_only _gazebo.launch.py`, just after line 62, I think should be added:
```
rrbot_forward_controller = os.path.join(
        get_package_share_directory("ros2_control_demo_robot"), "config", "rrbot_controllers.yaml"
    )

    spawn_controller = Node(
        package="controller_manager",
        executable="spawner.py",
        arguments=["forward_position_controller", "--param-file", rrbot_forward_controller, "-t", "forward_command_controller/ForwardCommandController"],
        output="screen",
    )
```
So that the rrbot Gazebo example can be run with `ros2 launch ros2_control_demo_robot  rrbot_system_position_only_gazebo.launch.py`

Let me know if there's something you'd like me to investigate further regarding this Github Issue. Thanks for your contributions!

bmagyar, 2021-06-03:
Could you please open a pull request with these changes?

olivier-stasse, 2021-06-22:
Dear @nicholaspalomo could you confirm that you still have a problem with the current HEAD of master, i.e. commit 3326857b6186772269e57d75d7da3b25ba9681a0 ?

nicholaspalomo, 2021-06-23:
> Dear @nicholaspalomo could you confirm that you still have a problem with the current HEAD of master, i.e. commit [3326857](https://github.com/ros-controls/ros2_control_demos/commit/3326857b6186772269e57d75d7da3b25ba9681a0) ?

Hi, @olivier-stasse. Sorry for my late reply. I'll check it now and get back to you.

nicholaspalomo, 2021-06-23:
> Hi, @olivier-stasse. Sorry for my late reply. I'll check it now and get back to you.

@olivier-stasse I started from a completely new/clean workspace. One thing I noticed is I get an error during the build process:
```
Could not find a package configuration file provided by "test_msgs" with
  any of the following names:

    test_msgsConfig.cmake
    test_msgs-config.cmake
...
```
When I tried installing `test_msgs` from binaries, it seems there was none available for `foxy`. I cloned `rcl_interfaces` to my `src` folder and checkout out the `foxy` branch. That resolved the issue for me. So, maybe in the `ros2_control_demos.repos` file the following lines:
```
  rcl_interfaces:
    type: git
    url: https://github.com/ros2/rcl_interfaces.git
    version: foxy
```
could be added.

I also tried running the RRbot Gazebo example.
```
ros2 launch ros2_control_demo_bringup rrbot_system_position_only_gazebo.launch.py
```
Unfortunately, it seems like the `joint_state_broadcaster` fails to spawn:
```
[spawner.py-5] [INFO] [1624473903.781432050] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [INFO] [1624473905.792608533] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [INFO] [1624473907.803833537] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [ERROR] [1624473909.813738332] [spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner.py-5]: process has died [pid 85081, exit code 1, cmd '/home/nicholas/workspace/dummy_ws/install/lib/controller_manager/spawner.py joint_state_broadcaster --ros-args'].
```
I'm not 100% sure why this is happening, especially since [I do almost the exact same thing](https://github.com/nicholaspalomo/panda_ros2_gazebo/blob/master/launch/gazebo.launch.py#L75) in a project of mine and the joint state broadcaster spawns without issue.

olivier-stasse, 2021-06-24:
@nicholaspalomo Thanks for the tests. I will investiguate.

Karsten1987, 2021-06-24:
> Could not find a package configuration file provided by "test_msgs"

That's not an issue with the demos. That's pretty normal if you compile your packages from source as their dependencies are not automatically installed opposed to a binary installation via `aptitude`.  For source builds you can crawl your workspace and install their declared dependencies as follows:
```
rosdep install --from-paths src --ignore-src --rosdistro=foxy -y
```

> When I tried installing test_msgs from binaries, it seems there was none available for foxy

The package is called `ros-foxy-test-msgs`. Note that underscores `_` are replaced with dashes `-`.

Once you've fixed your installation, can you please try a fresh clean build of your workspace and verify that the rrbot demo still does not work?

blakermchale, 2021-07-10:
I tried the rrbot gazebo demo also and had the same issue where the controller manager did not spawn. I didn't have any issues during the build process either. Here's a snippet from my terminal:
```
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_link had 1 children
[robot_state_publisher-3] Link link1 had 1 children
[robot_state_publisher-3] Link link2 had 3 children
[robot_state_publisher-3] Link camera_link had 1 children
[robot_state_publisher-3] Link camera_link_optical had 0 children
[robot_state_publisher-3] Link hokuyo_link had 0 children
[robot_state_publisher-3] Link tool_link had 0 children
[robot_state_publisher-3] [INFO] [1625896402.774375148] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1625896402.774428218] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-3] [INFO] [1625896402.774433127] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-3] [INFO] [1625896402.774436804] [robot_state_publisher]: got segment hokuyo_link
[robot_state_publisher-3] [INFO] [1625896402.774460509] [robot_state_publisher]: got segment link1
[robot_state_publisher-3] [INFO] [1625896402.774464115] [robot_state_publisher]: got segment link2
[robot_state_publisher-3] [INFO] [1625896402.774467472] [robot_state_publisher]: got segment tool_link
[robot_state_publisher-3] [INFO] [1625896402.774470848] [robot_state_publisher]: got segment world
[spawner.py-5] [INFO] [1625896402.887729461] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawn_entity.py-4] [INFO] [1625896403.000861777] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1625896403.001097300] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] [INFO] [1625896403.001862636] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1625896403.017304300] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1625896403.017605235] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1625896403.771084996] [spawn_entity]: Calling service /spawn_entity
[gzserver-1] [INFO] [1625896404.126696396] [camera_controller]: Publishing camera info to [/camera1/camera_info]
[spawn_entity.py-4] [INFO] [1625896404.131884184] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [rrbot_system_position]
[gzserver-1] [WARN] [1625896404.155856850] [gazebo_ros_head_hokuyo_controller]: missing <output_type>, defaults to sensor_msgs/PointCloud2
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 69130]
[spawner.py-5] [INFO] [1625896404.895066548] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [INFO] [1625896406.903291762] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [INFO] [1625896408.910918844] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [INFO] [1625896410.918821286] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [ERROR] [1625896412.926279155] [spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner.py-5]: process has died [pid 69133, exit code 1, cmd '/opt/ros/foxy/lib/controller_manager/spawner.py joint_state_broadcaster --ros-args'].
```

destogl, 2021-07-10:
@blakermchale thanks for pointing this out. I'll look at this later today.

destogl, 2023-02-02:
obsolete.

closed 2023-02-02 by destogl