ros-controls/ros2_control_demos #98: Error compiling packages with colcon build --symlink-install

robofoundry, 2021-06-03:
After cloning the repos and following instructions in read me, I get compile error, output from console below. Any help or pointers greatly appreciated. My installation is for ros foxy.

Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
Finished <<< ros2_control_test_assets [0.57s]                                                                    
Starting >>> hardware_interface
Finished <<< hardware_interface [1.09s]                                                                            
Starting >>> controller_interface
Finished <<< controller_interface [0.71s]                                                                          
Starting >>> ros2_control_test_nodes
Finished <<< controller_manager_msgs [2.95s]                                                                
Starting >>> controller_manager
Finished <<< ros2_control_test_nodes [2.00s]                                                             
Starting >>> ros2_control_demo_hardware
--- stderr: ros2_control_demo_hardware                                                                            
In file included from /home/ubuntu/dev/ros_control_demos/ros2_control_demos/ros2_control_demo_hardware/src/rrbot_system_multi_interface.cpp:15:
/home/ubuntu/dev/ros_control_demos/ros2_control_demos/ros2_control_demo_hardware/include/ros2_control_demo_hardware/rrbot_system_multi_interface.hpp:53:15: error: ‘hardware_interface::return_type ros2_control_demo_hardware::RRBotSystemMultiInterfaceHardware::prepare_command_mode_switch(const std::vector<std::__cxx11::basic_string<char> >&, const std::vector<std::__cxx11::basic_string<char> >&)’ marked ‘override’, but does not override
   53 |   return_type prepare_command_mode_switch(
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/ros2_control_demo_hardware.dir/build.make:89: CMakeFiles/ros2_control_demo_hardware.dir/src/rrbot_system_multi_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/ros2_control_demo_hardware.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ros2_control_demo_hardware [5.68s, exited with code 2]
Aborted  <<< controller_manager [12.6s]                                  

Summary: 5 packages finished [16.0s]
  1 package failed: ros2_control_demo_hardware
  1 package aborted: controller_manager
  2 packages had stderr output: controller_manager ros2_control_demo_hardware
  14 packages not processed

olivier-stasse, 2021-06-03:
Did you check if your computer has not a ros2_control binary package installed ?
My experience is that trying to compile  ros2_control from sources together with the binary packages will not work. There is a change in the API.

robofoundry, 2021-06-03:
@olivier-stasse - you are absolutely correct. I do have binary packages installed for ros2_control. I'll try to uninstall and try again. Thanks so much for the pointer, very much appreciate it. For future though what is the best way to run the demos and still keep updated with latest binaries, are there any best practice recommendations for the community?

bmagyar, 2021-06-06:
We try to keep things released as much as possible but one of our latest releases had a bundle of small breaking changes. This may have caused the issue you saw, sorry about it. This is not standard practice so the latest binaries should typically work for you however. If the latest demos repo doesn't compile it is because many things are being actively developed and the sync from the build farm did not yet happen.

Tldr: always try using binaries. If that doesn't work, grab all the sources or wait for the next release.

robofoundry, 2021-06-06:
Thank you so much Bence for commenting and caring!! This is what makes ROS community so awesome. I'll keep your tip in mind and get latest binaries and try the demos again.

olivier-stasse, 2021-06-09:
@robofoundry Thanks for your appreciation. @bmagyar is doing an impressive job on this project.
Could you close the issue if you are satisfied ?
Thanks a lot.

closed 2021-06-09 by robofoundry

newcanopies, 2021-06-11:
hi @olivier-stasse 

would such a binary vs source conflict raise this error at running the demo:
$ `ros2 launch ros2_control_demo_robot rrbot_system_position_only.launch.py`
```
[rviz2-3] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
```

would un-installing debian control packages be the only workaround?

Kind thanks!