ros-controls/ros2_control_demos #99: Diff Drive Demo not loading diff_drive_controller

tonybaltovski, 2021-06-03:
I am trying to run the diff drive demo but the `diff_drive_controller` doesn't load.  I run `ros2 launch ros2_control_demo_robot diffbot_system.launch.py` and it start successfully.   Checking the loaded controllers with `ros2 control list_controllers`, only `joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active` is shown.  I can load it manually but it stays inactive.  Any insights to how to get the controller loading on start?

olivier-stasse, 2021-06-10:
Dear @tonybaltovski Thanks for the report.
Could you confirm that you have the following exception in the terminal where you launched the demo:
`[ros2_control_node-1] ros2_control_node: /opt/ros/foxy/include/rcppmath/clamp.hpp:41: constexpr const T& rcppmath::clamp(const T&, const T&, const T&) [with T = double]: Assertion `!(hi < lo)' failed.` ?
I run into a similar problem and found a possible fix. But I still need to track down the root of the problem.

tonybaltovski, 2021-06-15:
@olivier-stasse that does not seem to be the issue.  The node does not crash.

tonybaltovski, 2021-06-22:
Pulling in the recent changes fixed the issue.  I do believe it could be an issue with name spacing in ros2_controllers but haven't got a chance to investigate it.

closed 2021-06-22 by tonybaltovski

olivier-stasse, 2021-06-22:
@tonybaltovski glad that the recent changes fixed the issue.