ros-controls/ros2_control_demos #107: [rviz2-3] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame

newcanopies, 2021-06-11:
$ `ros2 launch ros2_control_demo_robot rrbot_system_position_only.launch.py`
```
[rviz2-3] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
```

what could be throwing this error?

destogl, 2021-06-12:
Can you try or #76? This error always comes, but it is not persistent. The rviz robot visualization causes it

Karsten1987, 2021-06-14:
TL;DR: Make sure the joint state broadcaster is loaded and running!

The error message is maybe not as clear as it could be, but its cause is relatively simple and straightforward. TF is being published by the robot state publisher - which takes in joint states, performs FK and publishes the kinematic chain in task space. Now, the reason the frame doesn't exist is because nobody publishes the joint states.
In the context of ros2_control you have to make sure that the joint_state_broadcaster, formerly known as joint_state_controller is loaded and actively running. That broadcaster is responsible for publishing the joint states from your ros2_control'ed hardware so that the RSP can publish tf.

newcanopies, 2021-06-16:
> Can you try or #76?

Today, pulling 76 breaks the launch:
  File "<frozen importlib._bootstrap_external>", line 779, in exec_module
  File "<frozen importlib._bootstrap_external>", line 915, in get_code
  File "<frozen importlib._bootstrap_external>", line 972, in get_data

Was there supposed to be another PR to try before letters _or_ 76 ?
I'm aware PR work is on-going, many thanks!

olivier-stasse, 2021-06-16:
Dear @newcanopies #76 has been merged to master.

Therefore `ros2_demo_control_robot` does not exist anymore.

You should use
`ros2 launch ros2_control_demo_bringup rrbot_system_position_only.launch.py `
instead.

newcanopies, 2021-06-16:
Dear @olivier-stasse 

I pulled the merged Master branch, symlink rebuilt and ran the command with following error:

```
[robot_state_publisher-2] [INFO] [1623849443.972490800] [robot_state_publisher]: got segment world
[ros2_control_node-1] [INFO] [1623849443.976244800] [RRBotSystemPositionOnlyHardware]: Starting ...please wait...
[spawner.py-4] [INFO] [1623849444.141214700] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [INFO] [1623849444.142010300] [spawner_forward_position_controller]: Waiting for /controller_manager services
[rviz2-3] [INFO] [1623849444.238352000] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1623849444.238554000] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[rviz2-3] [INFO] [1623849444.260247000] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] Parsing robot urdf xml string.
[rviz2-3] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
```

My use case requires both Rviz **and Gazebo**, so I also tested:
$ `ros2 launch ros2_control_demo_bringup rrbot_system_position_only_gazebo.launch.py`
```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [1291]
[INFO] [gzclient   -2]: process started with pid [1293]
[INFO] [ros2_control_node-3]: process started with pid [1296]
[INFO] [robot_state_publisher-4]: process started with pid [1299]
[INFO] [rviz2-5]: process started with pid [1301]
[INFO] [spawner.py-6]: process started with pid [1303]
[INFO] [spawner.py-7]: process started with pid [1305]
[INFO] [spawn_entity.py-8]: process started with pid [1307]
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] Link base_link had 1 children
[robot_state_publisher-4] Link link1 had 1 children
[robot_state_publisher-4] Link link2 had 3 children
[robot_state_publisher-4] Link camera_link had 1 children
[robot_state_publisher-4] Link camera_link_optical had 0 children
[robot_state_publisher-4] Link hokuyo_link had 0 children
[robot_state_publisher-4] Link tool_link had 0 children
[robot_state_publisher-4] [INFO] [1623849357.283296000] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1623849357.283410900] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-4] [INFO] [1623849357.283430600] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-4] [INFO] [1623849357.283440400] [robot_state_publisher]: got segment hokuyo_link
[robot_state_publisher-4] [INFO] [1623849357.283448900] [robot_state_publisher]: got segment link1
[robot_state_publisher-4] [INFO] [1623849357.283457900] [robot_state_publisher]: got segment link2
[robot_state_publisher-4] [INFO] [1623849357.283467200] [robot_state_publisher]: got segment tool_link
[robot_state_publisher-4] [INFO] [1623849357.283475700] [robot_state_publisher]: got segment world
[ros2_control_node-3] [INFO] [1623849357.293212300] [controller_manager]: update rate is 2 Hz
[ros2_control_node-3] [INFO] [1623849357.508231400] [controller_manager]: Loading controller 'joint_state_broadcaster'
[rviz2-5] [INFO] [1623849357.624541800] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1623849357.624805000] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[rviz2-5] [INFO] [1623849357.650657300] [rviz2]: Stereo is NOT SUPPORTED
[spawner.py-6] [INFO] [1623849357.796982400] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-3] [INFO] [1623849357.799786100] [controller_manager]: Loading controller 'forward_position_controller'
[rviz2-5] Parsing robot urdf xml string.
[rviz2-5] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[spawn_entity.py-8] [INFO] [1623849357.839777300] [spawn_entity]: Spawn Entity started
[spawn_entity.py-8] [INFO] [1623849357.840153200] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-8] [INFO] [1623849357.843224000] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-8] [INFO] [1623849357.857309200] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-8] [INFO] [1623849357.857899300] [spawn_entity]: Waiting for service /spawn_entity
[rviz2-5] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[... 97 lines cut ...]
[... cut ...]
[rviz2-5] [INFO] [1623849383.261512700] [rclcpp]: signal_handler(signal_value=2)
[robot_state_publisher-4] [INFO] [1623849383.261501400] [rclcpp]: signal_handler(signal_value=2)
[ros2_control_node-3] [INFO] [1623849383.261562300] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-4]: process has finished cleanly [pid 1299]
[INFO] [ros2_control_node-3]: process has finished cleanly [pid 1296]
[INFO] [rviz2-5]: process has finished cleanly [pid 1301]
```

am I missing a step?
Thank you!

olivier-stasse, 2021-06-16:
> Dear @olivier-stasse
> 
> I pulled the merged Master branch, symlink rebuilt, sourced install, and ran the command w/ following error:
> $ `ros2 launch ros2_control_demo_bringup rrbot_system_position_only.launch.py`
> 
> Rviz loads just a blank white window with logo
> 
> ```
> [robot_state_publisher-2] [INFO] [1623849443.972490800] [robot_state_publisher]: got segment world
> [ros2_control_node-1] [INFO] [1623849443.976244800] [RRBotSystemPositionOnlyHardware]: Starting ...please wait...
> [spawner.py-4] [INFO] [1623849444.141214700] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
> [spawner.py-5] [INFO] [1623849444.142010300] [spawner_forward_position_controller]: Waiting for /controller_manager services
> [rviz2-3] [INFO] [1623849444.238352000] [rviz2]: Stereo is NOT SUPPORTED
> [rviz2-3] [INFO] [1623849444.238554000] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
> [rviz2-3] [INFO] [1623849444.260247000] [rviz2]: Stereo is NOT SUPPORTED
> [rviz2-3] Parsing robot urdf xml string.
> [rviz2-3] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
> [rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
> [rviz2-3] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
> [rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
> ```
Sorry maybe I missed it, but I do not see an error rather warnings. Could you highlight the error you are talking about ?

For the second part of your discussion, could you elaborate on which version gazebo_ros2_control and gazebo_ros_pkgs you are working ? I tested everything with commit 1b90ab1c4f2830 for gazebo_ros2_control and gazebo_ros_pkgs version 3.5.3 (commit 3f18465d3f)

olivier-stasse, 2021-06-16:
Dear @newcanopies I forgot to mention but indeed they are some problems with gazebo right now on master.
You could try to apply #109 .

olivier-stasse, 2021-06-22:
Dear @newcanopies the new changes should have fixed your issue of having rviz being blank. This will not fix the warnings. The gazebo simulation have also been fixed.

newcanopies, 2021-06-28:
Thank you kindly @olivier-stasse, the error seems to have changed:

> #76 has been merged to master.
> 
> Therefore `ros2_demo_control_robot` does not exist anymore.
> 
> You should use
> `ros2 launch ros2_control_demo_bringup rrbot_system_position_only.launch.py `
> instead.

Above `position_only` works after pulling today's Master sources for rps2_control, ros2_controllers and ros2_control_demos.

However, in the Gazebo launch:

`ros2 launch ros2_control_demo_bringup rrbot_system_position_only_gazebo.launch.py`

now the joint_state_broadcaster in controller_manager `spawner.py-5` process dies with the following error seen at the very end:

<details>
```
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2021-06-28-16-32-10-680044-machine-20121
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [20124]
[INFO] [gzclient   -2]: process started with pid [20126]
[INFO] [robot_state_publisher-3]: process started with pid [20129]
[INFO] [spawn_entity.py-4]: process started with pid [20132]
[INFO] [spawner.py-5]: process started with pid [20134]
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_link had 1 children
[robot_state_publisher-3] Link link1 had 1 children
[robot_state_publisher-3] Link link2 had 3 children
[robot_state_publisher-3] Link camera_link had 1 children
[robot_state_publisher-3] Link camera_link_optical had 0 children
[robot_state_publisher-3] Link hokuyo_link had 0 children
[robot_state_publisher-3] Link tool_link had 0 children
[robot_state_publisher-3] [INFO] [1624890731.434654800] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1624890731.434796200] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-3] [INFO] [1624890731.434818000] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-3] [INFO] [1624890731.434829400] [robot_state_publisher]: got segment hokuyo_link
[robot_state_publisher-3] [INFO] [1624890731.434839900] [robot_state_publisher]: got segment link1
[robot_state_publisher-3] [INFO] [1624890731.434849600] [robot_state_publisher]: got segment link2
[robot_state_publisher-3] [INFO] [1624890731.434859100] [robot_state_publisher]: got segment tool_link
[robot_state_publisher-3] [INFO] [1624890731.434868200] [robot_state_publisher]: got segment world
[spawner.py-5] [INFO] [1624890731.595598700] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawn_entity.py-4] [INFO] [1624890731.811206800] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1624890731.811601600] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] [INFO] [1624890731.813444600] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1624890731.843861800] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1624890731.844180100] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1624890732.599951300] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1624890733.109068100] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [rrbot_system_position]
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 20132]
[gzserver-1] [INFO] [1624890733.228168600] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1624890733.234685100] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1624890733.235103200] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1624890733.235150400] [gazebo_ros2_control]: Loading parameter file /home/esofi/ws_control/install/ros2_control_demo_bringup/share/ros2_control_demo_bringup/config/rrbot_gazebo_forward_controller_position.yaml
[gzserver-1]
[gzserver-1] [INFO] [1624890733.240254200] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1624890733.241337900] [gazebo_ros2_control]: Recieved urdf from param server, parsing...
[gzserver-1] [INFO] [1624890733.255886200] [gazebo_ros2_control]: Loading joint: joint1
[gzserver-1] [INFO] [1624890733.255978300] [gazebo_ros2_control]:       Command:
[gzserver-1] [INFO] [1624890733.255992500] [gazebo_ros2_control]:
position
[gzserver-1] [INFO] [1624890733.256000900] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1624890733.256007400] [gazebo_ros2_control]:
position
[gzserver-1] [INFO] [1624890733.256017000] [gazebo_ros2_control]: Loading joint: joint2
[gzserver-1] [INFO] [1624890733.256041800] [gazebo_ros2_control]:       Command:
[gzserver-1] [INFO] [1624890733.256047800] [gazebo_ros2_control]:
position
[gzserver-1] [INFO] [1624890733.256053700] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1624890733.256059700] [gazebo_ros2_control]:
position
[gzserver-1] [INFO] [1624890733.259310900] [gazebo_ros2_control]: Loading controller_manager
[gzserver-1] gzserver: symbol lookup error: /opt/ros/foxy/lib/libgazebo_ros2_control.so: undefined symbol: _ZN18controller_manager17ControllerManagerC1ESt10unique_ptrIN18hardware_interface15ResourceManagerESt14default_deleteIS3_EESt10shared_ptrIN6rclcpp8ExecutorEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
[... cut ...]
[spawner.py-5] [INFO] [1624890739.675902500] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [ERROR] [1624890741.699860600] [spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner.py-5]: process has died [pid 20134, exit code 1, cmd '/home/user/ws_control/install/controller_manager/lib/controller_manager/spawner.py joint_state_broadcaster --ros-args'].
```

control workspace is upto date with Master, dependencies installed, rebult and sourced.

newcanopies, 2021-06-28:
> make sure that the joint_state_broadcaster, formerly known as joint_state_controller is loaded and actively running.

Presumably the demo rrbot joint_state_broadcaster is loaded by the Gazebo launch file:

https://github.com/ros-controls/ros2_control_demos/blob/bf32f38d4d3dca02bde7d46ac15af51fe4870c66/ros2_control_demo_bringup/launch/rrbot_system_position_only_gazebo.launch.py#L62-L67

Karsten1987, 2021-06-28:
ros2_control in Foxy had ABI breaking changes. We ran into the same problem the other day. In order to fix your problem you have essentially two options:

* Change your official ROS apt repository to use the testing repository. http://wiki.ros.org/TestingRepository#Usage
* Wait until the next Foxy sync in which all testing repositories are matched with the official release repository

destogl, 2023-02-02:
obosolete...

closed 2023-02-02 by destogl