ros-controls/ros2_control_demos #115: Example 4 : Add code for the demo example

bailaC, 2021-06-21:
[Example - 4](https://github.com/ros-controls/ros2_control_demos#example-4-differential-drive-mobile-robot)

I will pick this item and start working on it.

bailaC, 2021-06-22:
@olivier-stasse I think this is already done at [here](https://github.com/ros-controls/ros2_control_demos/pull/113)

@destogl can you please review once? If so, we can close this as duplicate.

olivier-stasse, 2021-06-22:
@bailaC Sorry I have mistaken example 4 (diff_robot) for example 3.

The PR you are mentionning was merged too quickly by me. 
@destogl disagre with some of the code style and I was planning to work on it by the end of the week. But the example is starting for me. 

Let me know if there is a problem because not it is affecting master.

bailaC, 2021-06-26:
@olivier-stasse I see many files are merged with their old code and coding styles as well. I am not sure what should be next move here. There is no problem so far. But I am not able to see any visible output of the PR-113.

olivier-stasse, 2021-06-26:
Dear @bailaC, some of the inconsistencies  has been fixed partially by PR #117.
They are some discussions on PR #118.

Example 4 has been added.

bailaC, 2021-06-28:
The [example-4](https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md#4-industrial-robots-with-externally-connected-sensor) is taken from here.

destogl, 2022-03-25:
This is done.

closed 2022-03-25 by destogl