ros-controls/ros2_control_demos #130: Example of loading controller from `controller_manger`

tonybaltovski, 2021-08-09:
I have written a `hardware_interface` for Husky and am able to load it from the command line.  However, I cannot load using the `controller_manager`.  Is there an example that better explains how to that?  Source for current work: https://github.com/husky/husky/tree/wip-ros2-control/husky_base

Karsten1987, 2021-08-09:
I am not sure I follow what exactly you're trying to load. The hardware interfaces are solely loaded through the resource manager and all declared state and command interfaces are made available through the controller manager.
You shouldn't really have to deal with the controller manager.

I guess the example around the `diffbot` could serve you as a decent entry point: https://github.com/ros-controls/ros2_control_demos/blob/master/ros2_control_demo_bringup/launch/diffbot_system.launch.py
Once you've launched this, you can introspect the system with the following CLI tools:

```
 ➭ ros2 control list_hardware_interfaces
command interfaces
	left_wheel_joint/velocity [claimed]
	right_wheel_joint/velocity [claimed]
state interfaces
	 left_wheel_joint/position
	 left_wheel_joint/velocity
	 right_wheel_joint/position
	 right_wheel_joint/velocity
```

```
 ➭ ros2 control list_controllers
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
diffbot_base_controller[diff_drive_controller/DiffDriveController] active
```

tonybaltovski, 2021-08-10:
Thanks! Awesome, it worked!

closed 2021-08-10 by tonybaltovski