ros-controls/ros2_control_demos #134: Could not contact service /controller_manager/list_controllers

Richard-Haes-Ellis, 2021-09-14:
I've followed the demo for the RRBot and all works fine until I try to list the controllers. I get this message:

`Could not contact service /controller_manager/list_controllers`

I also get the same message when calling:
```
- ros2 control list_hardware_interfaces
- ros2 control list_controller_types
- ros2 control list_controllers
```

I haven't tested the rest of the services but I'm guessing that the controller_manager isn't online or not launched properly. 

Furthermore I've also tried to run a ros2_control launch file for ODrive with a basic velocity controller (https://github.com/Factor-Robotics/odrive_ros2_control.git) and I also get a related error from the controller_manager. It is unable to contact the service.  
```
ros2 launch odrive_bringup odrive.launch.py       
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2021-09-14-16-40-07-155005-ubuntu-2816
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [2820]
[INFO] [robot_state_publisher-2]: process started with pid [2822]
[INFO] [spawner.py-3]: process started with pid [2824]
[INFO] [spawner.py-4]: process started with pid [2826]
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] Link link0 had 0 children
[robot_state_publisher-2] [INFO] [1631637608.196999177] [robot_state_publisher]: got segment link0
[robot_state_publisher-2] [INFO] [1631637608.197494333] [robot_state_publisher]: got segment world
[spawner.py-3] Traceback (most recent call last):
[spawner.py-3]   File "/opt/ros/foxy/lib/controller_manager/spawner.py", line 186, in <module>
[spawner.py-3]     sys.exit(main())
[spawner.py-3]   File "/opt/ros/foxy/lib/controller_manager/spawner.py", line 109, in main
[spawner.py-3]     if is_controller_loaded(node, controller_manager_name, controller_name):
[spawner.py-3]   File "/opt/ros/foxy/lib/controller_manager/spawner.py", line 51, in is_controller_loaded
[spawner.py-3]     controllers = list_controllers(node, controller_manager).controller
[spawner.py-3]   File "/opt/ros/foxy/lib/python3.8/site-packages/controller_manager/controller_manager_services.py", line 49, in list_controllers
[spawner.py-3]     return service_caller(node, f'{controller_manager_name}/list_controllers',
[spawner.py-3]   File "/opt/ros/foxy/lib/python3.8/site-packages/controller_manager/controller_manager_services.py", line 29, in service_caller
[spawner.py-3]     raise RuntimeError(f'Could not contact service {service_name}')
[spawner.py-3] RuntimeError: Could not contact service /controller_manager/list_controllers
[spawner.py-4] Traceback (most recent call last):
[spawner.py-4]   File "/opt/ros/foxy/lib/controller_manager/spawner.py", line 186, in <module>
[spawner.py-4]     sys.exit(main())
[spawner.py-4]   File "/opt/ros/foxy/lib/controller_manager/spawner.py", line 109, in main
[spawner.py-4]     if is_controller_loaded(node, controller_manager_name, controller_name):
[spawner.py-4]   File "/opt/ros/foxy/lib/controller_manager/spawner.py", line 51, in is_controller_loaded
[spawner.py-4]     controllers = list_controllers(node, controller_manager).controller
[spawner.py-4]   File "/opt/ros/foxy/lib/python3.8/site-packages/controller_manager/controller_manager_services.py", line 49, in list_controllers
[spawner.py-4]     return service_caller(node, f'{controller_manager_name}/list_controllers',
[spawner.py-4]   File "/opt/ros/foxy/lib/python3.8/site-packages/controller_manager/controller_manager_services.py", line 29, in service_caller
[spawner.py-4]     raise RuntimeError(f'Could not contact service {service_name}')
[spawner.py-4] RuntimeError: Could not contact service /controller_manager/list_controllers
[ERROR] [spawner.py-3]: process has died [pid 2824, exit code 1, cmd '/opt/ros/foxy/lib/controller_manager/spawner.py joint_state_broadcaster --controller-manager /controller_manager --ros-args'].
[ERROR] [spawner.py-4]: process has died [pid 2826, exit code 1, cmd '/opt/ros/foxy/lib/controller_manager/spawner.py joint0_velocity_controller -c /controller_manager --ros-args'].
```
Looking at discussions about this error I found a discussion talking about the stability of the spawn.py of the controller_manager 
https://github.com/ros-controls/ros2_control/issues/475

They mention it could be because it's running on a slower PC but I'm running the same code on raspberry pi and my laptop PC with powerful specs that shouldn't be a problem and yet both platforms yield the same result. 

I assume it's a bug but have no clue how to debug this issue. Any help is deeply appreciated.

bmagyar, 2021-09-15:
Not sure what you are running but I'd try removing the spawners from your main launch file and first letting the rest start up, then inspecting if the controller_manager services are available. 

The spawner script has a [timeout](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/scripts/spawner.py#L89) you could change but the default is 10s which I think is fairly generous...

Richard-Haes-Ellis, 2021-09-16:
I've figured out the culprit, it seems that the hardware_interface hangs waiting for my device to respond and messes up the controller_manager start-up. I'll close. Thanks!

closed 2021-09-16 by Richard-Haes-Ellis

bmagyar, 2021-09-17:
That however sir may be something we are able to assess very soon! @destogl here's another example of why we need component lifecycles :D

Gilocho, 2022-02-26:
> I've figured out the culprit, it seems that the hardware_interface hangs waiting for my device to respond and messes up the controller_manager start-up. I'll close. Thanks!

apparently I have the same problem as you, how did you solve it?, I'm a bit confused with this, thanks for the help in advance

T2FF, 2022-03-22:
> I've figured out the culprit, it seems that the hardware_interface hangs waiting for my device to respond and messes up the controller_manager start-up. I'll close. Thanks!

Hi i also have the same problem, is there a way to solve the hardware interface problem?

omr-web, 2022-08-24:
I have the same problem but I dont understand how can ı solve it. Could you help me please

rda11, 2022-10-23:
> I've figured out the culprit, it seems that the hardware_interface hangs waiting for my device to respond and messes up the controller_manager start-up. I'll close. Thanks!

can you please explain a little more...

unusmani786, 2024-03-21:
can anyone please list the answer

christophfroehlich, 2024-03-22:
> can anyone please list the answer

Have you tried the suggestion here? https://github.com/ros-controls/ros2_control_demos/issues/134#issuecomment-920390955
As this is a rather old thread, the code changed and improved a lot: It might not be the same issue as the original one at foxy.

GradVizor, 2024-10-04:
can anyone please help me with this, I'm having these error lines:-
"[spawner-4] [INFO] [1728064935.284322938] [spawner_r6bot_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-4] [WARN] [1728064945.310500069] [spawner_r6bot_controller]: Could not contact service /controller_manager/list_controllers". 
I've tried everything, even I tried source building it but getting new errors in that way, so I reverted back to this, but still not getting any positive feedback.

SKAAAN47, 2024-10-13:
> can anyone please help me with this, I'm having these error lines:- "[spawner-4] [INFO] [1728064935.284322938] [spawner_r6bot_controller]: waiting for service /controller_manager/list_controllers to [...]

did you find a solution ?

SKAAAN47, 2024-10-21:
> can anyone please help me with this, I'm having these error lines:- "[spawner-4] [INFO] [1728064935.284322938] [spawner_r6bot_controller]: waiting for service /controller_manager/list_controllers to [...]

I still have the problem for more than  a week now and it seems their is no answer listed anywhere about this kind of problem

saikishor, 2024-10-21:
Hello!
Can you share some logs of what controller_manager is printing?

SKAAAN47, 2024-10-21:
Hello,
Thanks for the quick answer.

spawn_entity.py-4] [INFO] [1729497684.636715683] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1729497684.872663143] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [milltap700_5x]
[gazebo-1] [INFO] [1729497684.978992734] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gazebo-1] [INFO] [1729497684.980592841] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gazebo-1] [INFO] [1729497684.980691824] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gazebo-1] [INFO] [1729497684.982108750] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gazebo-1] [INFO] [1729497684.982462460] [gazebo_ros2_control]: Received urdf from param server, parsing...
[gazebo-1] [INFO] [1729497684.982937163] [gazebo_ros2_control]: Loading parameter files /home/skan/ws2/install/robotpkg/share/robotpkg/config/milltap700_5x.yaml
[gazebo-1] [ERROR] [1729497684.983070021] [gazebo_ros2_control]: parser error Couldn't parse params file: '--params-file /home/skan/ws2/install/robotpkg/share/robotpkg/config/milltap700_5x.yaml'. Error: Cannot have a value before ros__parameters at line 3, at ./src/parse.c:793, at ./src/rcl/arguments.c:406
[gazebo-1] 
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 12221]
[gazebo-1] [Msg] Waiting for master.
[gazebo-1] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gazebo-1] [Msg] Publicized address: 192.168.2.118
[gazebo-1] [Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[spawner-6] [WARN] [1729497693.825393465] [spawner_milltap_controller]: Could not contact service /controller_manager/list_controllers
[spawner-6] [INFO] [1729497693.826121086] [spawner_milltap_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-5] [WARN] [1729497693.830148353] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers

Chillhopper, 2025-01-14:
Does anyone have a solution im going crazy

saikishor, 2025-01-14:
> Does anyone have a solution im going crazy

Make sure that the Hardware components are properly configured. Unless they are configured, you don't have the CM services

Chillhopper, 2025-01-14:
> Make sure that the Hardware components are properly configured. Unless they are configured, you don't have the CM services

Sorry by that do you mean <hardwareInterface>?

harshithadev, 2025-01-28:
> I've figured out the culprit, it seems that the hardware_interface hangs waiting for my device to respond and messes up the controller_manager start-up. I'll close. Thanks!

can you please explain more into this ? I seem to be experiencing the same!

Chillhopper, 2025-01-29:
> > I've figured out the culprit, it seems that the hardware_interface hangs waiting for my device to respond and messes up the controller_manager start-up. I'll close. Thanks!
> 
> can you please explain more into this ? I seem to be experiencing the same!

Hey there! So from what I know if you are getting this error its probebobly something to do with your hardware interfaces or ros2 control tags. Here is a link to my working urdf, you can compare 1 to 1 and see what's missing or what's different:

https://github.com/Chillhopper/ROS2_Elevator_Simulation/blob/main/description/my_robot.urdf.xacro

sggutierrez95, 2025-05-12:
My issue when I had this was to include
```py
    ros2_controllers_path = os.path.join(
        get_package_share_directory("robot_moveit_config"),
        "config",
        "ros2_controllers.yaml",
    )

    ros2_control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[ros2_controllers_path],
        remappings=[
            ("/controller_manager/robot_description", "/robot_description"),
        ],
        output="log",
    )

```
in my launch file

Hyz24dsdas, 2025-11-17:
https://github.com/legubiao/quadruped_ros2_control/issues/3