ros-controls/ros2_control_demos #153: diffbot not circuling in Rviz

HusamZain, 2022-03-09:
I'm using ROS2 galactic , and trying to do the diffbot example.
I downloaded ros2-control and ros2-controllers from apt not source 
everything seems to work fine, until when I try to write a message to /cmd_vel  as in step 5 nothing happen and the robot in rviz doesnt move 
[image]

I checked the topics , and this is what I found 
[image]
so I wrote to the /diffbot_base_controller/cmd_vel_unstamped , 
`ros2 topic pub --rate 30 diffbot_base_controller/cmd_vel_unstamped  geometry_msgs/msg/Twist "linear:
 x: 0.7
 y: 0.0
 z: 0.0
angular:
 x: 0.0
 y: 0.0
 z: 1.0"`

and the robot started circuiling, so is that means the command from you wrong and it should be `diffbot_base_controller/cmd_vel_unstamped ` instead of `/cmd_vel` ? or I did something wrong ?

destogl, 2022-03-09:
You did all good. Probably the documentation is not on the newest state. Can you please open PR with your fix?

HusamZain, 2022-03-10:
@destogl 
Thank you for your feedback! 
Yes sure, I'm new to github, so not sure if I did it right !

closed 2022-03-12 by HusamZain

be-glass, 2022-04-09:
It would be very nice to place a warning in the documentation. It took me two 2 days of debugging before I found this.