ros-controls/ros2_control_demos #180: undefined symbol: builtin_interfaces__msg__Duration__are_equal

GeorgeZZZZZ, 2022-05-20:
Hi, I'm using binary installed galactic, this error has not shown before the update few days ago. I'm new to ros so I have no Idea how to fix this, I try to reinstall ros2 package and reinstall ubuntu to get a fresh install but still get this error. I don't know what to do now, may be I should delete all packages and build from source? any suggestion, thanks.

`[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [30193]
[INFO] [robot_state_publisher-2]: process started with pid [30195]
[INFO] [spawner-3]: process started with pid [30197]
[robot_state_publisher-2] Link base_link had 1 children
[robot_state_publisher-2] Link link1 had 1 children
[robot_state_publisher-2] Link link2 had 3 children
[robot_state_publisher-2] Link camera_link had 1 children
[robot_state_publisher-2] Link camera_link_optical had 0 children
[robot_state_publisher-2] Link hokuyo_link had 0 children
[robot_state_publisher-2] Link tool_link had 0 children
[robot_state_publisher-2] [INFO] [1653053501.669724974] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1653053501.669776195] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-2] [INFO] [1653053501.669783425] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-2] [INFO] [1653053501.669788245] [robot_state_publisher]: got segment hokuyo_link
[robot_state_publisher-2] [INFO] [1653053501.669793025] [robot_state_publisher]: got segment link1
[robot_state_publisher-2] [INFO] [1653053501.669797725] [robot_state_publisher]: got segment link2
[robot_state_publisher-2] [INFO] [1653053501.669802345] [robot_state_publisher]: got segment tool_link
[robot_state_publisher-2] [INFO] [1653053501.669806795] [robot_state_publisher]: got segment world
[ros2_control_node-1] [INFO] [1653053501.672330876] [RRBotSystemPositionOnlyHardware]: Configuring ...please wait...
[ros2_control_node-1] [INFO] [1653053501.672373436] [RRBotSystemPositionOnlyHardware]: Successfully configured!
[ros2_control_node-1] [INFO] [1653053501.672573938] [RRBotSystemPositionOnlyHardware]: Activating ...please wait...
[ros2_control_node-1] [INFO] [1653053501.672579318] [RRBotSystemPositionOnlyHardware]: Successfully activated!
[ros2_control_node-1] [INFO] [1653053501.675487761] [controller_manager]: update rate is 10 Hz
[ros2_control_node-1] [INFO] [1653053501.775739212] [RRBotSystemPositionOnlyHardware]: Reading...
[ros2_control_node-1] [INFO] [1653053501.775787262] [RRBotSystemPositionOnlyHardware]: Got state 0.00000 for joint 0!
[ros2_control_node-1] [INFO] [1653053501.775803192] [RRBotSystemPositionOnlyHardware]: Got state 0.00000 for joint 1!
[ros2_control_node-1] [INFO] [1653053501.775810582] [RRBotSystemPositionOnlyHardware]: Joints successfully read!
[ros2_control_node-1] [INFO] [1653053501.775839933] [RRBotSystemPositionOnlyHardware]: Writing...
[ros2_control_node-1] [INFO] [1653053501.775847423] [RRBotSystemPositionOnlyHardware]: Got command 0.00000 for joint 0!
[ros2_control_node-1] [INFO] [1653053501.775855643] [RRBotSystemPositionOnlyHardware]: Got command 0.00000 for joint 1!
[ros2_control_node-1] [INFO] [1653053501.775863143] [RRBotSystemPositionOnlyHardware]: Joints successfully written!
[spawner-3] Traceback (most recent call last):
[spawner-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
[spawner-3]     return importlib.import_module(module_name, package=pkg_name)
[spawner-3]   File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
[spawner-3]     return _bootstrap._gcd_import(name[level:], package, level)
[spawner-3]   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
[spawner-3]   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
[spawner-3]   File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
[spawner-3]   File "<frozen importlib._bootstrap>", line 657, in _load_unlocked
[spawner-3]   File "<frozen importlib._bootstrap>", line 556, in module_from_spec
[spawner-3]   File "<frozen importlib._bootstrap_external>", line 1166, in create_module
[spawner-3]   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
[spawner-3] ImportError: /mnt/hp_500g/0.Projects/ros_projects/ros2_control_demos/install/controller_manager_msgs/lib/libcontroller_manager_msgs__rosidl_generator_c.so: undefined symbol: builtin_interfaces__msg__Duration__are_equal
[spawner-3] 
[spawner-3] During handling of the above exception, another exception occurred:
[spawner-3] 
[spawner-3] Traceback (most recent call last):
[spawner-3]   File "/mnt/hp_500g/0.Projects/ros_projects/ros2_control_demos/install/controller_manager/lib/controller_manager/spawner", line 20, in <module>
[spawner-3]     ret = main()
[spawner-3]   File "/mnt/hp_500g/0.Projects/ros_projects/ros2_control_demos/src/ros2_control-galactic/controller_manager/scripts/spawner.py", line 167, in main
[spawner-3]     if is_controller_loaded(node, controller_manager_name, controller_name):
[spawner-3]   File "/mnt/hp_500g/0.Projects/ros_projects/ros2_control_demos/src/ros2_control-galactic/controller_manager/scripts/spawner.py", line 110, in is_controller_loaded
[spawner-3]     controllers = list_controllers(node, controller_manager).controller
[spawner-3]   File "/mnt/hp_500g/0.Projects/ros_projects/ros2_control_demos/install/controller_manager/lib/python3.8/site-packages/controller_manager/controller_manager_services.py", line 49, in list_controllers
[spawner-3]     return service_caller(node, f'{controller_manager_name}/list_controllers',
[spawner-3]   File "/mnt/hp_500g/0.Projects/ros_projects/ros2_control_demos/install/controller_manager/lib/python3.8/site-packages/controller_manager/controller_manager_services.py", line 23, in service_caller
[spawner-3]     cli = node.create_client(service_type, service_name)
[spawner-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 1409, in create_client
[spawner-3]     check_is_valid_srv_type(srv_type)
[spawner-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/type_support.py", line 51, in check_is_valid_srv_type
[spawner-3]     check_for_type_support(srv_type)
[spawner-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
[spawner-3]     msg_or_srv_type.__class__.__import_type_support__()
[spawner-3]   File "/mnt/hp_500g/0.Projects/ros_projects/ros2_control_demos/install/controller_manager_msgs/lib/python3.8/site-packages/controller_manager_msgs/srv/_list_controllers.py", line 250, in __import_type_support__
[spawner-3]     module = import_type_support('controller_manager_msgs')
[spawner-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
[spawner-3]     raise UnsupportedTypeSupport(pkg_name)
[spawner-3] rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'controller_manager_msgs'
`

bmagyar, 2022-05-20:
Which branch of demos are you using?

Also...this linking error seems pretty significant, I'm afraid your email
have a broken setup but don't know how on a mint system...

GeorgeZZZZZ, 2022-05-21:
Using galactic branch. 
Turns out I add mirco-roc source in bashrc when I was testing codes on esp32. Now I remove source every thing works fine.
Thanks for the reply anyway.

closed 2022-05-21 by GeorgeZZZZZ

bmagyar, 2022-05-21:
We have CI building stuff in a variety of setups, it's just extremely
unlikely anyone approaching us with a build error is not coming with an
issue that's their mistake ;)