ros-controls/ros2_control_demos #186: parameter 'joints' has already been declared

ijnek, 2022-06-13:
When following instructions in README.md and running:

```
ros2 launch ros2_control_demo_bringup rrbot.launch.py
```

An exception is thrown, with the forward_position_controller failing to start.

```
[ros2_control_node-1] Exception thrown during init stage with message: parameter 'joints' has already been declared 
[ros2_control_node-1] [ERROR] [1655080823.709643215] [controller_manager]: Could not initialize the controller named 'forward_position_controller'
```

Then, running ``ros2 control list_controllers`` as in the README.md returns:

```
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
```

Missing ``forward_position_controller[forward_command_controller/ForwardCommandController] active``

pierre-rouanet, 2022-06-23:
It was fixed there I think: https://github.com/ros-controls/ros2_controllers/commit/547cef784df4fda35cebd6a1c1ab282d7b24f83d

ijnek, 2022-06-23:
Great, thanks @pierre-rouanet

closed 2022-06-23 by ijnek