ros-controls/ros2_control_demos #209: change hw_state_ to hw_position for only position command rrbot.

benthebear93, 2022-09-17:
As a beginner, i been reading code for rrbot and diffbot together.

One thing that made me little confusing was variable's name. 

variable for state of position only rrbot is 

`std::vector<double> hw_states_;`

variable for diffbot is

```
  std::vector<double> hw_positions_;
  std::vector<double> hw_velocities_;

```
and rrbot_system_multi_interface has

```
  std::vector<double> hw_states_positions_;
  std::vector<double> hw_states_velocities_;
  std::vector<double> hw_states_accelerations_;
```

it's trivial but wouldn't it be better to unify the names of variables?

bmagyar, 2022-09-17:
Such suggestions are best communicated through PRs ;)

benthebear93, 2022-09-21:
@bmagyar I will think about more general ways to unify the variables name and do the PR.

destogl, 2022-12-13:
@benthebear93 are you working on this, or should we close the issue?

destogl, 2023-01-19:
Closing since it is stale...

closed 2023-01-19 by destogl