ros-controls/ros2_control_demos #231: Strategy for bringing the repository up to date

destogl, 2023-02-02:
This issue provides an overview and steps to bring the repository up to the speed of the rest of the framework.

**Goal**: Get all existing examples into new structure, working with the master branches of `ros2_control` and `ros2_controllers` repositories, and make maintenance easy.

Check #

Let's add all the example to the above PR and merge the whole restructuring at once. For that, we will squash-merge all the PR expect the last one to the master.

# Target Structure and Usage of Functionalities from Other Repositories

- [ ] Add each example into separate self-contained package
- [ ] Unify naming of the files are proposed in the PR
- [ ] Use .rst files for describing examples and the main repository description to be easily integrated into documentation.
  - ros-controls/ros2_control_demos#105
- [ ] Use `CMakeLists.txt` structure for overriding showed in - ros_controls/ros2_control#926

# Examples

- [ ] Example_1 - *RRBot* - single interface robot arm example (merged with "Single interface industrial robot example")
- [ ] Example_2 - *DiffBot* - single interface mobile robot example
- [ ] Example_3 – "Robots with multiple interfaces"
- [ ] Example_4 – "Industrial robot with integrated sensor"
- [ ] Example_5 – "Industrial Robots with externally connected sensor"
- [ ] Example_6 – "Modular Robots with separate communication to each actuator"
- [ ] … TBA …

christophfroehlich, 2023-02-15:
should we add the [mode-switching](https://github.com/ros-controls/ros2_control_demos/blob/master/doc/index.rst#mode-switching-demo) as a standalone example, or add this to example_1?

christophfroehlich, 2023-02-19:
> should we add the [mode-switching](https://github.com/ros-controls/ros2_control_demos/blob/master/doc/index.rst#mode-switching-demo) as a standalone example, or add this to example_1?

I propose integrating it into example_1 with #243 , because the controller-yaml already had the JTC config.

christophfroehlich, 2023-02-21:
I had a look on the [images ](https://github.com/ros-controls/ros2_control_demos/tree/master/doc)in the doc folder, which are outdated imho. @destogl can we update them or should we delete them?

PR #232 (2023-03-08): Restructuring of the repository for better overview.
This PR introduces the restructured version of the repository.

Each example should be submitted as separate PR to this branch.
All examples should be merged at once.

Resolves #231.

Before merging this PR we should at least do the following:

- [x] rename `use_fake_hardware` to `use_mock_hardware`
- [ ] Remove `rrbot_base.launch.py` from example folders, i.e., rename it to example launch file and setup parameters needed for the example.
- [x] #240 
- [x] remove use of **NOTE**: from .rst files and use .. note:: block
- [x] Use CMakeLists.t [... cut]