ros-controls/ros2_control_demos #251: ros2_control with ethercat_driver_ros2 errors when running ros2_control demo with ethercat driver plugin

NamanMo97, 2023-03-08:
I am trying to find a way to have a connection between ros2_control and ethercat_driver. the master is Activated but still get the error even when defining the ec_module as a plugin, the module is uploaded but still got the same error I defined the ethercat driver as a plugin in a new description file which is system_description.ros2_control.xacro and called it from the main rrbot_system_multi_interface.urdf.xacro description file, what I am trying to do is to make a description file with ethercat driver and ethercat module plugins in order to be launched from the rrbot_system_multi_interface.launch.py in the demos package,
**is there any specified informations about how to use ros2_control with ethercat_driver?** here are the decription files and the launch file that I am using:

**rrbot_system_multi_interface.urdf.xacro**

```
<?xml version="1.0"?>
<!-- Revolute-Revolute Manipulator -->
<!--
Copied and modified from ROS1 example -
https://github.com/ros-simulation/gazebo_ros_demos/blob/kinetic-devel/rrbot_description/urdf/rrbot.xacro
-->
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="2dof_robot">
  <xacro:arg name="use_gazebo" default="false" />

  <!-- Enable setting arguments from the launch file -->
  <xacro:arg name="use_fake_hardware" default="false" />
  <xacro:arg name="mock_sensor_commands" default="false" />
  <xacro:arg name="prefix" default="" />
  <xacro:arg name="slowdown" default="100.0" />

  <!-- Import RRBot macro -->
  <xacro:include filename="$(find rrbot_description)/urdf/rrbot_description.urdf.xacro" />

  <!-- Import all Gazebo-customization elements, including Gazebo colors -->
  <xacro:include filename="$(find rrbot_description)/gazebo/rrbot.gazebo.xacro" />

  <!-- Import Rviz colors -->
  <xacro:include filename="$(find rrbot_description)/gazebo/rrbot.materials.xacro" />

  <!-- Import RRBot ros2_control description -->
  <xacro:include filename="$(find rrbot_description)/ros2_control/rrbot_system_multi_interface.ros2_control.xacro" />
  <xacro:include filename="$(find rrbot_description)/ros2_control/system_description.ros2_control.xacro" />

  <!-- Used for fixing robot -->
  <link name="world"/>
  <gazebo reference="world">
    <static>true</static>
  </gazebo>

  <xacro:rrbot parent="world" prefix="$(arg prefix)">
    <origin xyz="0 0 0" rpy="0 0 0" />
  </xacro:rrbot>

  <xacro:rrbot_gazebo prefix="$(arg prefix)" />

  <xacro:rrbot_system_multi_interface
    name="RRBotSystemMultiInterface" prefix="$(arg prefix)"
    use_fake_hardware="$(arg use_fake_hardware)"
    mock_sensor_commands="$(arg mock_sensor_commands)"
    slowdown="$(arg slowdown)" />

</robot>
```

**system_description.ros2_control.xacro**

```
<?xml version="1.0"?>
<!-- Revolute-Revolute Manipulator -->
<!--
Copied and modified from ROS1 example -
https://github.com/ros-simulation/gazebo_ros_demos/blob/kinetic-devel/rrbot_description/urdf/rrbot.xacro
-->
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="2dof_robot">

  <ros2_control name="mySystem" type="system">
    <hardware>
        <plugin>ethercat_driver/EthercatDriver</plugin>
        <param name="master_id">0</param>
        <param name="control_frequency">100</param>
    </hardware>
  </ros2_control> 
 
</robot>
```

**rrbot_system_multi_interface.launch.py**

```
# Copyright 2021 Department of Engineering Cybernetics, NTNU.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration, ThisLaunchFileDir

def generate_launch_description():
    # Declare arguments
    declared_arguments = []
    declared_arguments.append(
        DeclareLaunchArgument(
            "prefix",
            default_value='""',
            description="Prefix of the joint names, useful for \
        multi-robot setup. If changed than also joint names in the controllers' configuration \
        have to be updated.",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument(
            "slowdown", default_value="50.0", description="Slowdown factor of the RRbot."
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument(
            "robot_controller",
            default_value="forward_velocity_controller",
            description="Robot controller to start.",
        )
    )

    # Initialize Arguments
    prefix = LaunchConfiguration("prefix")
    slowdown = LaunchConfiguration("slowdown")
    robot_controller = LaunchConfiguration("robot_controller")

    base_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([ThisLaunchFileDir(), "/rrbot_base.launch.py"]),
        launch_arguments={
            "controllers_file": "rrbot_multi_interface_forward_controllers.yaml",
            "description_file": "rrbot_system_multi_interface.urdf.xacro",
            "prefix": prefix,
            "use_fake_hardware": "false",
            "mock_sensor_commands": "false",
            "slowdown": slowdown,
            "robot_controller": robot_controller,
        }.items(),
    )

    return LaunchDescription(declared_arguments + [base_launch])

```

christophfroehlich, 2023-03-13:
Hi! I never worked with ethercat and ros2, but as written in its [readme](https://github.com/ICube-Robotics/ethercat_driver_ros2) it is designed for ros2_control -> there has to be a way to get that running and I don't think its an issue of the demo in this repository. But maybe @mcbed can give you an answer here.

mcbed, 2023-03-17:
Hi @NamanMo97, 
As you probably already did, you need to add to the `ros2_control` urdf description the ec_module that you want to use for each joint/gpio component using:
```xml
<joint name="myJoint">
  <command_interface name="xxx"/>
  <state_interface name="xxx"/>
  <ec_module name="ECModule">
    <plugin>ethercat_plugins/ECModule</plugin>
    <param name="alias">xx</param>
    <param name="position">xx</param>
  </ec_module>
</joint>
```
I would need some more info about your setup to help you out setting it up. I suggest we move this topic to the [ethercat_driver_ros2](https://github.com/ICube-Robotics/ethercat_driver_ros2) repo.

christophfroehlich, 2023-03-17:
> I want to ask you if you have used the **ros2_control_demos** package to have an implementation of **ros2_control** on a real robot?

What do you mean exactly? Yes, I used these hardware interfaces as templates for a real robot.

mcbed, 2023-03-17:
> > I want to ask you if you have used the **ros2_control_demos** package to have an implementation of **ros2_control** on a real robot?
> 
> What do you mean exactly? Yes, I used these hardware interfaces as templates for a real robot.

Same for us, new `ros2_control` hardware integration mostly starts from this repo.

NamanMo97, 2023-03-17:
@christophfroehlich @mcbed 
For my question I just wanted to check if I'm starting my integration correctly.
I will move this topic to the [ethercat_driver_ros2](https://github.com/ICube-Robotics/ethercat_driver_ros2) repo with more details as @mcbed suggested.