ros-controls/ros2_control_demos #261: example_1 :

olivier-stasse, 2023-03-26:
Dear @christophfroehlich, 
Thanks again for the great work in restructuring the repo.

After cloning ```ros2_control``` from HEAD to [c8cfa05](https://github.com/ros-controls/ros2_control/commit/c8cfa053de8a2d69d6f10e770fefbd4ff051d58a), ```ros2_controllers``` from HEAD to  82a4e24 

I tried the following:
```
ros2 launch ros2_control_demo_example_1 rrbot.launch.py
ros2 launch ros2_control_demo_example_1 test_joint_trajectory_controller.launch.py
```
and the robot is not moving. 

The output of ```rrbot.launch.py```:
```
[ros2_control_node-1] [INFO] [1679797583.074655377] [RRBotSystemPositionOnlyHardware]: Reading...
[ros2_control_node-1] [INFO] [1679797583.074724709] [RRBotSystemPositionOnlyHardware]: Got state 0.00000 for joint 0!
[ros2_control_node-1] [INFO] [1679797583.074769125] [RRBotSystemPositionOnlyHardware]: Got state 0.00000 for joint 1!
[ros2_control_node-1] [INFO] [1679797583.074802707] [RRBotSystemPositionOnlyHardware]: Joints successfully read!
[ros2_control_node-1] [INFO] [1679797583.074926205] [RRBotSystemPositionOnlyHardware]: Writing...
[ros2_control_node-1] [INFO] [1679797583.074956538] [RRBotSystemPositionOnlyHardware]: Got command 0.00000 for joint 0!
[ros2_control_node-1] [INFO] [1679797583.074993370] [RRBotSystemPositionOnlyHardware]: Got command 0.00000 for joint 1!
[ros2_control_node-1] [INFO] [1679797583.075030203] [RRBotSystemPositionOnlyHardware]: Joints successfully written!
[ros2_control_node-1] [INFO] [1679797583.175206210] [RRBotSystemPositionOnlyHardware]: Reading...
[ros2_control_node-1] [INFO] [1679797583.175299375] [RRBotSystemPositionOnlyHardware]: Got state 0.00000 for joint 0!
[ros2_control_node-1] [INFO] [1679797583.175336207] [RRBotSystemPositionOnlyHardware]: Got state 0.00000 for joint 1!
[ros2_control_node-1] [INFO] [1679797583.175373040] [RRBotSystemPositionOnlyHardware]: Joints successfully read!
[ros2_control_node-1] [INFO] [1679797583.175533370] [RRBotSystemPositionOnlyHardware]: Writing...
[ros2_control_node-1] [INFO] [1679797583.175573453] [RRBotSystemPositionOnlyHardware]: Got command 0.00000 for joint 0!
[ros2_control_node-1] [INFO] [1679797583.175660118] [RRBotSystemPositionOnlyHardware]: Got command 0.00000 for joint 1!
[ros2_control_node-1] [INFO] [1679797583.175700200] [RRBotSystemPositionOnlyHardware]: Joints successfully written!
[ros2_control_node-1] [INFO] [1679797583.275331301] [RRBotSystemPositionOnlyHardware]: Reading...
[ros2_control_node-1] [INFO] [1679797583.275404967] [RRBotSystemPositionOnlyHardware]: Got state 0.00000 for joint 0!
[ros2_control_node-1] [INFO] [1679797583.275446132] [RRBotSystemPositionOnlyHardware]: Got state 0.00000 for joint 1!
[ros2_control_node-1] [INFO] [1679797583.275498131] [RRBotSystemPositionOnlyHardware]: Joints successfully read!
[ros2_control_node-1] [INFO] [1679797583.275955289] [RRBotSystemPositionOnlyHardware]: Writing...
[ros2_control_node-1] [INFO] [1679797583.276004038] [RRBotSystemPositionOnlyHardware]: Got command 0.00000 for joint 0!
[ros2_control_node-1] [INFO] [1679797583.276044121] [RRBotSystemPositionOnlyHardware]: Got command 0.00000 for joint 1!
[ros2_control_node-1] [INFO] [1679797583.276079870] [RRBotSystemPositionOnlyHardware]: Joints successfully written!

```
The output of ```publisher_joint_trajectory_controller```:
```
[publisher_joint_trajectory_controller-1] <goal_name>:
[publisher_joint_trajectory_controller-1]   positions: [joint1, joint2, joint3, ...]
[publisher_joint_trajectory_controller-1]   velocities: [v_joint1, v_joint2, ...]
[publisher_joint_trajectory_controller-1]   accelerations: [a_joint1, a_joint2, ...]
[publisher_joint_trajectory_controller-1]   effort: [eff_joint1, eff_joint2, ...]
[publisher_joint_trajectory_controller-1] [INFO] [1679797544.706363868] [publisher_joint_trajectory_controller]: Publishing 4 goals on topic "/position_trajectory_controller/joint_trajectory" every {wait_sec_between_publish} s
[publisher_joint_trajectory_controller-1] [INFO] [1679797550.731068065] [publisher_joint_trajectory_controller]: Sending goal trajectory_msgs.msg.JointTrajectoryPoint(positions=[0.785, 0.785], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0)).
[publisher_joint_trajectory_controller-1] [INFO] [1679797556.730399966] [publisher_joint_trajectory_controller]: Sending goal trajectory_msgs.msg.JointTrajectoryPoint(positions=[0.0, 0.0], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0)).
[publisher_joint_trajectory_controller-1] [INFO] [1679797562.731126088] [publisher_joint_trajectory_controller]: Sending goal trajectory_msgs.msg.JointTrajectoryPoint(positions=[-0.785, -0.785], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0)).
[publisher_joint_trajectory_controller-1] [INFO] [1679797568.732063456] [publisher_joint_trajectory_controller]: Sending goal trajectory_msgs.msg.JointTrajectoryPoint(positions=[0.0, 0.0], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0)).
[publisher_joint_trajectory_controller-1] [INFO] [1679797574.731105029] [publisher_joint_trajectory_controller]: Sending goal trajectory_msgs.msg.JointTrajectoryPoint(positions=[0.785, 0.785], velocities=[], accelerations=[], effort=[], time_from_start=builtin_interfaces.msg.Duration(sec=4, nanosec=0)).
```

Do you have any hints on what I am missing ?

christophfroehlich, 2023-03-26:
have you activated `joint_trajectory_controller`? per default, only the `forward_command_controller` is active. See point 6 in the [docs](https://control.ros.org/master/doc/ros2_control_demos/example_1/doc/userdoc.html)

olivier-stasse, 2023-03-26:
Excellent ! Thanks for the fast answer. IHMO maybe the file README.md could be update to point towards the doc
I am closing the issue.

closed 2023-03-26 by olivier-stasse

christophfroehlich, 2023-03-26:
You mean instead of the link to the folder only? Maybe we should add a simple readme in every folder with the only content "Documentation can be found in doc/userdoc.rst" ;)

olivier-stasse, 2023-03-26:
> You mean instead of the link to the folder only? Maybe we should add a simple readme in every folder with the only content "Documentation can be found in doc/userdoc.rst" ;)

Your link was extremely useful. I do not get why I missed it in the documentation. Maybe putting the equivalent link in each README.md would be very useful.