ros-controls/ros2_control_demos #285: colcon build issue - galactic

Jdhaworth, 2023-05-15:
**Describe the bug**
After cloning all of the necessary packages (for the galactic branch) and running vsc import and rosdep I build using:
```
 colcon build --symlink-install
```
During the build I receive the following errors:
```
--- stderr: control_msgs                                                                        
CMake Error: The source "/home/jdhaworthrsp/rsp_fp_ws/src/control_msgs/control_msgs/CMakeLists.txt" does not match the source "/home/jdhaworthrsp/rsp_fp_ws/src/ros-controls/control_msgs/control_msgs/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
---
Failed   <<< control_msgs [1.20s, exited with code 1]
--- stderr: realtime_tools
CMake Error: The source "/home/jdhaworthrsp/rsp_fp_ws/src/realtime_tools/CMakeLists.txt" does not match the source "/home/jdhaworthrsp/rsp_fp_ws/src/ros-controls/realtime_tools/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
---
Failed   <<< realtime_tools [1.18s, exited with code 1]
```
Any idea what may be causing this? Thanks for your time.

christophfroehlich, 2023-05-16:
Removing the build folder often fixes such problems (rm -rf build).

christophfroehlich, 2023-05-16:
Read the error message again, it states that you have different paths. have you cloned the repository twice? If you have manually cloned the "packages"(repos), then you don't have to call vcs import any more.

Jdhaworth, 2023-05-16:
I think that fixed it. Thanks!