ros-controls/ros2_control_demos #344: Build error due to wrong brnach in tutorial.

kaltenegger, 2023-08-16:
**Describe the bug**
Build of `ros2_control_demo_example_12` fails.

**To Reproduce**
Pull commit from master branch and run `colcon build --merge-install` (according to the `ros2_control` [tutorial](https://control.ros.org/master/doc/ros2_control_demos/doc/index.html)).
This results in the following output containing the error:
```
➜  ros2_control_demos colcon build --merge-install
Starting >>> ros2_control_demo_example_1
Starting >>> ros2_control_demo_example_2
Starting >>> ros2_control_demo_example_3
Starting >>> ros2_control_demo_example_4
Starting >>> ros2_control_demo_example_5
Starting >>> ros2_control_demo_example_6
Starting >>> ros2_control_demo_example_8
Starting >>> ros2_control_demo_example_9
Starting >>> ros2_control_demo_example_12
Starting >>> ros2_control_demo_example_7
Finished <<< ros2_control_demo_example_3 [0.28s]                                    
Finished <<< ros2_control_demo_example_2 [0.30s]                                   
Finished <<< ros2_control_demo_example_4 [0.30s]
Finished <<< ros2_control_demo_example_1 [0.34s]
Finished <<< ros2_control_demo_example_6 [0.34s]
Finished <<< ros2_control_demo_example_5 [0.35s]
Finished <<< ros2_control_demo_example_9 [0.35s]
Finished <<< ros2_control_demo_example_8 [0.36s]
Starting >>> ros2_control_demos
Finished <<< ros2_control_demo_example_7 [0.39s]                                      
Finished <<< ros2_control_demos [0.09s]
--- stderr: ros2_control_demo_example_12                               
In file included from /home/eugen/Projects/ROS2/ros2_control_demos/src/ros2_control_demos/example_12/controllers/src/passthrough_controller.cpp:15:
/home/eugen/Projects/ROS2/ros2_control_demos/src/ros2_control_demos/example_12/controllers/include/passthrough_controller/passthrough_controller.hpp:74:37: error: ‘controller_interface::return_type passthrough_controller::PassthroughController::update_reference_from_subscribers(const rclcpp::Time&, const rclcpp::Duration&)’ marked ‘override’, but does not override
   74 |   controller_interface::return_type update_reference_from_subscribers(
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp:57,
                 from /opt/ros/humble/include/class_loader/class_loader/class_loader.hpp:55,
                 from /opt/ros/humble/include/pluginlib/pluginlib/class_list_macros.hpp:40,
                 from /home/eugen/Projects/ROS2/ros2_control_demos/src/ros2_control_demos/example_12/controllers/src/passthrough_controller.cpp:17:
/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::impl::MetaObject<C, B>::create() const [with C = passthrough_controller::PassthroughController; B = controller_interface::ChainableControllerInterface]’:
/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:216:7:   required from here
/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:218:12: error: invalid new-expression of abstract class type ‘passthrough_controller::PassthroughController’
  218 |     return new C;
      |            ^~~~~
In file included from /home/eugen/Projects/ROS2/ros2_control_demos/src/ros2_control_demos/example_12/controllers/src/passthrough_controller.cpp:15:
/home/eugen/Projects/ROS2/ros2_control_demos/src/ros2_control_demos/example_12/controllers/include/passthrough_controller/passthrough_controller.hpp:40:7: note:   because the following virtual functions are pure within ‘passthrough_controller::PassthroughController’:
   40 | class PassthroughController : public controller_interface::ChainableControllerInterface
      |       ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/eugen/Projects/ROS2/ros2_control_demos/src/ros2_control_demos/example_12/controllers/include/passthrough_controller/passthrough_controller.hpp:24,
                 from /home/eugen/Projects/ROS2/ros2_control_demos/src/ros2_control_demos/example_12/controllers/src/passthrough_controller.cpp:15:
/opt/ros/humble/include/controller_interface/chainable_controller_interface.hpp:100:23: note:     ‘virtual controller_interface::return_type controller_interface::ChainableControllerInterface::update_reference_from_subscribers()’
  100 |   virtual return_type update_reference_from_subscribers() = 0;
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/passthrough_controller.dir/build.make:76: CMakeFiles/passthrough_controller.dir/controllers/src/passthrough_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:193: CMakeFiles/passthrough_controller.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros2_control_demo_example_12 [2.96s, exited with code 2]

Summary: 10 packages finished [3.11s]
  1 package failed: ros2_control_demo_example_12
  1 package had stderr output: ros2_control_demo_example_12
```
Switching to `humble` branch solves the issue, this should be mentioned in the tutorial. 

**Expected behavior**
Build without any errors.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04.3
 - Version: Humble

**Additional context**
Solution proposal: modifying the tutorials to provide the command to clone the proper branch (e.g.: humble) or give a hint, that the default branch is not the current LTS.

christophfroehlich, 2023-08-16:
Hi! the master branch of this repo is targeting rolling distro (see, e.g., [here](https://github.com/ros-controls/ros2_control_demos#build-status). But I agree that it could be written explicitly written [in the docs](https://github.com/ros-controls/ros2_control_demos#build-status).

kaltenegger, 2023-08-16:
Renaming the `master` branch to `rolling would allow to use `git clone --branch $ROS_DISTRO ...`. I think this would be a nice and clean solution.

christophfroehlich, 2023-08-16:
This is not the practice of ros-controls organization, but I made a PR clarifying this.

PR #345 (2023-08-17): Add info about ROS 2 distro
Improve docs by adding explicit info about the ROS 2 distro, close #344.

rst for rolling will be rendered this way:
[image]
closed 2023-08-17 by christophfroehlich