ros-controls/ros2_control_demos #381: launch error

Mark9999999, 2023-11-18:
**Describe the bug**
ros2 launch controlko_bringup rrbot.launch.py 
[INFO] [launch]: All log files can be found below /home/mark/.ros/log/2023-11-18-10-51-14-714783-mark-desktop-29875
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable '[<launch.substitutions.text_substitution.TextSubstitution object at 0xffffa4d32200>]' not found on the PATH

when i run this command on arm64 computer ,there is the error, and how to solve the ERROR?

christophfroehlich, 2023-11-18:
Hi! I don't think this is related to this repo. It seems that there is an issue with your ROS installation.
Which ROS distro are you using, and which operating system?

Mark9999999, 2023-11-18:
> Hi! I don't think this is related to this repo. It seems that there is an issue with your ROS installation. Which ROS distro are you using, and which operating system?

Thanks your reply
ros2 humble and ubuntu 22.04 
but  it can work, when I run this program in computer of x86

christophfroehlich, 2023-11-19:
Have you installed ROS from binaries on your arm platform? Have you tried to launch other launch files with success?
A [similar error](https://robotics.stackexchange.com/questions/103001/ros2-launch-gives-an-error-in-ros2-control-demos) was reported earlier, where some dependencies were the issue.

christophfroehlich, 2024-05-07:
Closing due to inactivity.

closed 2024-05-07 by christophfroehlich