ros-controls/ros2_control_demos #382: Renaming of examples

Nils-ChristianIseke, 2023-12-05:
I would prefer it if one could deduce the purpose of the example from the name of the example.
My first suggestion, would look like that: 
    1_Position_Control_RRBot
    2_Diff_Drive_DiffBot
    3_Multi_Interfaces_RRBot
    4_Integrated_Sensor_RRBot
    5_External_Sensor_RRBot
    6_Individual_Actuator_Communication_Modular_Robot
    7_6DOF_Robot
    8_Transmission_Interface_RRBot
    9_Gazebo_Sim_Classic_[what is the robot used?]
    10_GPIO_Interfaces_RRBot
    12_Chaining_Controllers_RRBot
    
If you have other naming ideas feel free to share them.

christophfroehlich, 2023-12-05:
We discussed this shortly when we reorganized this repository. But back then it was easier to just give them numbers. 

If we rename it, I'd rename the folders and package names equally.
Can ROS packages start with a number? I don't think so: https://ros.org/reps/rep-0144.html#id11

christophfroehlich, 2023-12-05:
But they are not really building on top of each other, maybe the numbers are not necessary at all.

Nils-ChristianIseke, 2023-12-05:
> We discussed this shortly when we reorganized this repository. But back then it was easier to just give them numbers.
> 
> If we rename it, I'd rename the folders and package names equally. Can ROS packages start with a number? I don't think so: https://ros.org/reps/rep-0144.html#id11
Yes right :p, my fault.

Not sure if numbers are necessary, i just kept them. I prefer no numbers at all :)

olivier-stasse, 2024-05-29:
Thanks for raising the question.

As a counter argument, I would argue that before we had no numbers and it was orderd alphabetically which is not good either because there is _some_  increasing level of difficulty.

This is not so well reflected in the numbering.  But coming back to a none numbered set of examples we will run into another presentation problem. So i would suggest a prefix: Ex_Nb which gives:
 ```
Ex_1_Position_Control_RRBot
Ex_2_Diff_Drive_DiffBot
Ex_3_Multi_Interfaces_RRBot
Ex_4_Integrated_Sensor_RRBot
Ex_5_External_Sensor_RRBot
Ex_6_Individual_Actuator_Communication_Modular_Robot
Ex_7_6DOF_Robot
Ex_8_Transmission_Interface_RRBot
Ex_9_Gazebo_Sim_Classic_RRBot
Ex_10_GPIO_Interfaces_RRBot
Ex_12_Chaining_Controllers_RRBot
 ```

christophfroehlich, 2024-08-02:
Fyi once #417 can be merged I'm open for renaming.
Some examples have the same robot and controllers, but only different hardware component structure (ex 1, 3, 6, 14). Maybe we can restructure that a bit and add Ex_1a, Ex_1b, etc. Any suggestions?

christophfroehlich, 2025-07-03:
Unfortunately, it never got "completed" because the discussion stopped.