ros-controls/ros2_control_demos #383: Enhancing README.md

Nils-ChristianIseke, 2023-12-05:
The README.md file does not provide enough support for inexperienced users. This causes unnecessary issues / work for / by inexperienced users.
The README should at least refer to the necessary resources, better still have a step by step guide on how to run the examples from scratch.

christophfroehlich, 2023-12-05:
The necessary information is inside the doc folder in rst syntax for https://control.ros.org/master/doc/ros2_control_demos/doc/index.html
I don't want to duplicate information actually.

But I see that there is a lack of an introduction in the readme.md pointing to the correct page on control.ros.org. Would this be sufficient?

Nils-ChristianIseke, 2023-12-05:
How about #384?

christophfroehlich, 2023-12-05:
closed with #384

closed 2023-12-05 by christophfroehlich