ros-controls/ros2_control_demos #390: Add rqt_joint_trajectory_controller to example_1

christophfroehlich, 2023-12-10:
## Background

Add the usage of rqt_joint_trajectory_controller to the demos, maybe included in example_1?

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control_demos/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/galactic/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control_demos`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control_demos` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control_demos repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/galactic/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

AdityaPawar162, 2025-03-06:
Hi @christophfroehlich can I claim this issue?

christophfroehlich, 2025-03-06:
Yess!

AdityaPawar162, 2025-03-07:
Hi @christophfroehlich,
I successfully ran example_1 using the rqt_trajectory_controller GUI. To do this, I first launched the rrbot in the first terminal. Then, in the second terminal, I loaded the joint_trajectory_position_controller, switched to the first controller, and finally ran the rqt_trajectory_controller GUI.

AdityaPawar162, 2025-03-08:
Hi @christophfroehlich,
I've added it to the documentation of Example 1 and am raising a PR.
Please take a look and let me know if you have any suggestions.

AdityaPawar162, 2025-03-08:
Hi @christophfroehlich,
I've added it to the documentation of Example 1 and am raising a PR(#734).
Please take a look and let me know if you have any suggestions.

christophfroehlich, 2025-03-11:
Fixed with #734

closed 2025-03-11 by christophfroehlich