ros-controls/ros2_control_demos #438: Getting error while building ros2_control_demo_example_8 package.

anubhav1772, 2024-01-26:
I am trying to build `ros2_control_demo_example_8` package but getting below error:
```
--- stderr: ros2_control_demo_example_8                                                                                      
/home/anubhav1772/ros2_ws/src/example_8/hardware/rrbot_transmissions_system_position_only.cpp: In member function ‘virtual hardware_interface::CallbackReturn ros2_control_demo_example_8::RRBotTransmissionsSystemPositionOnlyHardware::on_init(const hardware_interface::HardwareInfo&)’:
/home/anubhav1772/ros2_ws/src/example_8/hardware/rrbot_transmissions_system_position_only.cpp:98:24: error: ‘const struct hardware_interface::JointInfo’ has no member named ‘state_interfaces’; did you mean ‘interfaces’?
   98 |       if (!(joint_info.state_interfaces.size() == 1 &&
      |                        ^~~~~~~~~~~~~~~~
      |                        interfaces
/home/anubhav1772/ros2_ws/src/example_8/hardware/rrbot_transmissions_system_position_only.cpp:99:24: error: ‘const struct hardware_interface::JointInfo’ has no member named ‘state_interfaces’; did you mean ‘interfaces’?
   99 |             joint_info.state_interfaces[0] == hardware_interface::HW_IF_POSITION &&
      |                        ^~~~~~~~~~~~~~~~
      |                        interfaces
/home/anubhav1772/ros2_ws/src/example_8/hardware/rrbot_transmissions_system_position_only.cpp:100:24: error: ‘const struct hardware_interface::JointInfo’ has no member named ‘command_interfaces’
  100 |             joint_info.command_interfaces.size() == 1 &&
      |                        ^~~~~~~~~~~~~~~~~~
/home/anubhav1772/ros2_ws/src/example_8/hardware/rrbot_transmissions_system_position_only.cpp:101:24: error: ‘const struct hardware_interface::JointInfo’ has no member named ‘command_interfaces’
  101 |             joint_info.command_interfaces[0] == hardware_interface::HW_IF_POSITION))
      |                        ^~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/ros2_control_demo_example_8.dir/build.make:76: CMakeFiles/ros2_control_demo_example_8.dir/hardware/rrbot_transmissions_system_position_only.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/ros2_control_demo_example_8.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

I already have `ros-humble-hardware-interface` installed, if that's the requirement.

Am I doing something wrong here? I really need some advice on this.

bmagyar, 2024-01-26:
You are trying to build the master branch on Humble. Try the humble branch ;)

closed 2024-01-26 by bmagyar

anubhav1772, 2024-01-26:
I cloned the humble branch. I re-verified it.

anubhav1772, 2024-01-26:
As per the [documentation](https://docs.ros.org/en/humble/p/hardware_interface/generated/structhardware__interface_1_1JointInfo.html#_CPPv4N18hardware_interface9JointInfoE), `state_interfaces` and `command_interfaces` are the members of `JointInfo`. So, I shouldn't get these errors.

anubhav1772, 2024-01-26:
I found the mistake. I had `ros2_control` package in my workspace which was definitely not cloned from humble branch. So, `ros2_control_demo_example_8` must be referring to the hardware interface of that package instead of `/opt/ros/humble/include/hardware_interface`. Thanks!