ros-controls/ros2_control_demos #455: Demo 14 broken?

fmauch, 2024-02-29:
**Describe the bug**
Without further investigating this it seems that demo14 is currently broken at least on the master branch.

**To Reproduce**
Run demo 14 as described in the [docs](https://control.ros.org/master/doc/ros2_control_demos/example_14/doc/userdoc.html)

````
ros2 launch ros2_control_demo_example_14 rrbot_modular_actuators_without_feedback_sensors_for_position_feedback.launch.py
```

Output:
```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-02-29-19-32-42-042374-jamy-33260                                                                                                                                                                                                               
[INFO] [launch]: Default logging verbosity is set to INFO                                                                                                                                                                                                                                                                   
[INFO] [ros2_control_node-1]: process started with pid [33264]                                                                                                                                                                                                                                                              
[INFO] [robot_state_publisher-2]: process started with pid [33266]                                                                                                                                                                                                                                                          
[INFO] [spawner-3]: process started with pid [33268]                                                                                                                                                                                                                                                                        
[robot_state_publisher-2] [INFO] [1709235162.259080771] [robot_state_publisher]: Robot initialized                                                                                                                                                                                                                          
[ros2_control_node-1] [INFO] [1709235162.296861203] [controller_manager]: Subscribing to '/robot_description' topic for robot description.                                                                                                                                                                                  
[ros2_control_node-1] [INFO] [1709235162.297108401] [controller_manager]: update rate is 100 Hz                                                                                                                                                                                                                             
[ros2_control_node-1] [WARN] [1709235162.297235058] [controller_manager]: Could not enable FIFO RT scheduling policy. Consider setting up your user to do FIFO RT scheduling. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details.                                        
[ros2_control_node-1] [INFO] [1709235162.307313025] [controller_manager]: Received robot description from topic.                                                                                                                                                                                                            
[ros2_control_node-1] [INFO] [1709235162.307854653] [resource_manager]: Loading hardware 'RRBotModularPositionSensorJoint1'                                                                                                                                                                                                 
[ros2_control_node-1] [INFO] [1709235162.309086054] [resource_manager]: Initialize hardware 'RRBotModularPositionSensorJoint1'                                                                                                                                                                                              
[ros2_control_node-1] [INFO] [1709235162.309315034] [RRBotSensorPositionFeedback]: Setting socket options.                                                                                                                                                                                                                  
[ros2_control_node-1] [INFO] [1709235162.309327406] [RRBotSensorPositionFeedback]: Binding to socket address.                                                                                                                                                                                                               
[ros2_control_node-1] [INFO] [1709235162.309417033] [resource_manager]: Successful initialization of hardware 'RRBotModularPositionSensorJoint1'                                                                                                                                                                            
[ros2_control_node-1] [INFO] [1709235162.309436681] [RRBotSensorPositionFeedback]: Listening for connection on port 23286.                                                                                                                                                                                                  
[ros2_control_node-1] [INFO] [1709235162.309587643] [resource_manager]: Loading hardware 'RRBotModularPositionSensorJoint2'                                                                                                                                                                                                 
[ros2_control_node-1] [INFO] [1709235162.309663852] [resource_manager]: Initialize hardware 'RRBotModularPositionSensorJoint2'                                                                                                                                                                                              
[ros2_control_node-1] [INFO] [1709235162.309700970] [RRBotSensorPositionFeedback]: Setting socket options.                                                                                                                                                                                                                  
[ros2_control_node-1] [INFO] [1709235162.309707412] [RRBotSensorPositionFeedback]: Binding to socket address.                                                                                                                                                                                                               
[ros2_control_node-1] [INFO] [1709235162.309759944] [resource_manager]: Successful initialization of hardware 'RRBotModularPositionSensorJoint2'                                                                                                                                                                            
[ros2_control_node-1] [INFO] [1709235162.309777021] [RRBotSensorPositionFeedback]: Listening for connection on port 23287.                                                                                                                                                                                                  
[ros2_control_node-1] [INFO] [1709235162.309808843] [resource_manager]: Loading hardware 'RRBotModularJoint1'                                                                                                                                                                                                               
[ros2_control_node-1] [INFO] [1709235162.310270555] [resource_manager]: Initialize hardware 'RRBotModularJoint1'                                                                                                                                                                                                            
[ros2_control_node-1] [INFO] [1709235162.310459729] [RRBotActuatorWithoutFeedback]: Trying to connect to port 23286.                                                                                                                                                                                                        
[... cut ...]
========= kinematics_interface =========
6b5d811e73d2f019af98988805160bba96b3940c
========= realtime_tools =========
8995b494d4a0f3b0655f1aa9bffb92d43a988c46
========= ros2_control =========
0711cd6274eace78dde90f56b7994e60edba7697
========= ros2_control_demos =========
044db05c476c1bd99132b511e491fc00c8eabd56
========= ros2_controllers =========
00172ab5dbd4948df78493e5acd40901a7227976
```

**Additional context**
I didn't investigate this all too much but did not want that to get lost on the way.

christophfroehlich, 2024-03-01:
I can't reproduce it with a clean build of `colcon build --packages-up-to ros2_control_demo_example_14`

saikishor, 2024-03-01:
I can't reproduce it as well. It works for me :)

fmauch, 2024-03-02:
OK, the problems seems indeed to come from my change in #456. Looks like when using the description topic it tries to send commands to the hardware before it is activated.

```
[ros2_control_node-1] [INFO] [1709235162.311241783] [RRBotActuatorWithoutFeedback]: Activating ...please wait...
[ros2_control_node-1] [INFO] [1709235162.311684397] [RRBotSensorPositionFeedback]: Receiving data
[ros2_control_node-1] [INFO] [1709235162.312043305] [RRBotSensorPositionFeedback]: Receiving data
[ros2_control_node-1] [INFO] [1709235162.317357018] [RRBotActuatorWithoutFeedback]: Writing command: nan
[ros2_control_node-1] [INFO] [1709235162.317383404] [RRBotActuatorWithoutFeedback]: Sending data command: nan
[ros2_control_node-1] [INFO] [1709235162.317411609] [RRBotActuatorWithoutFeedback]: Joints successfully written!
```

vs. using the description parameter:
```
[ros2_control_node-1] [INFO] [1709406595.828694734] [RRBotActuatorWithoutFeedback]: Activating ...please wait...                                                                                                                                                                                                            
[spawner-3] [INFO] [1709406596.132883688] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available                                                                                                                                                                                     
[ros2_control_node-1] [INFO] [1709406596.828821648] [RRBotActuatorWithoutFeedback]: 2.0 seconds left...                                                                                                                                                                                                                     
[ros2_control_node-1] [INFO] [1709406597.829122935] [RRBotActuatorWithoutFeedback]: 1.0 seconds left...
[ros2_control_node-1] [INFO] [1709406597.829185501] [RRBotActuatorWithoutFeedback]: Successfully activated!
[ros2_control_node-1] [INFO] [1709406597.829215398] [resource_manager]: Successful 'activate' of hardware 'RRBotModularJoint1'
[ros2_control_node-1] [INFO] [1709406597.829412333] [resource_manager]: 'configure' hardware 'RRBotModularPositionSensorJoint2' 
[ros2_control_node-1] [INFO] [1709406597.829475210] [RRBotSensorPositionFeedback]: Configuration successful.
[ros2_control_node-1] [INFO] [1709406597.829491110] [resource_manager]: Successful 'configure' of hardware 'RRBotModularPositionSensorJoint2'
[ros2_control_node-1] [INFO] [1709406597.829526538] [resource_manager]: 'activate' hardware 'RRBotModularPositionSensorJoint2' 
[ros2_control_node-1] [INFO] [1709406597.829540464] [RRBotSensorPositionFeedback]: Activating ...please wait...
[spawner-3] [INFO] [1709406598.160180152] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[ros2_control_node-1] [INFO] [1709406598.829744724] [RRBotSensorPositionFeedback]: 1.0 seconds left...
[ros2_control_node-1] [INFO] [1709406598.829808242] [RRBotSensorPositionFeedback]: Successfully activated!
[ros2_control_node-1] [INFO] [1709406598.829836718] [resource_manager]: Successful 'activate' of hardware 'RRBotModularPositionSensorJoint2'
[ros2_control_node-1] [INFO] [1709406598.829935389] [resource_manager]: 'configure' hardware 'RRBotModularPositionSensorJoint1' 
[ros2_control_node-1] [INFO] [1709406598.829959936] [RRBotSensorPositionFeedback]: Configuration successful.
[ros2_control_node-1] [INFO] [1709406598.829973430] [resource_manager]: Successful 'configure' of hardware 'RRBotModularPositionSensorJoint1'
[ros2_control_node-1] [INFO] [1709406598.830003075] [resource_manager]: 'activate' hardware 'RRBotModularPositionSensorJoint1' 
[ros2_control_node-1] [INFO] [1709406598.830015991] [RRBotSensorPositionFeedback]: Activating ...please wait...
[ros2_control_node-1] [INFO] [1709406599.830308578] [RRBotSensorPositionFeedback]: 1.0 seconds left...
[ros2_control_node-1] [INFO] [1709406599.830371781] [RRBotSensorPositionFeedback]: Successfully activated!
[ros2_control_node-1] [INFO] [1709406599.830399869] [resource_manager]: Successful 'activate' of hardware 'RRBotModularPositionSensorJoint1'
[ros2_control_node-1] [INFO] [1709406599.842852174] [controller_manager]: update rate is 100 Hz
[ros2_control_node-1] [WARN] [1709406599.842956793] [controller_manager]: Could not enable FIFO RT scheduling policy. Consider setting up your user to do FIFO RT scheduling. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details.
[ros2_control_node-1] [INFO] [1709406599.843048536] [RRBotSensorPositionFeedback]: Reading...
[ros2_control_node-1] [INFO] [1709406599.843066684] [RRBotSensorPositionFeedback]: Got measured velocity nan
[ros2_control_node-1] [INFO] [1709406599.843072846] [RRBotSensorPositionFeedback]: Got state 0.00000 for joint 'joint1'!
[ros2_control_node-1] [INFO] [1709406599.843078447] [RRBotSensorPositionFeedback]: Joints successfully read!
[ros2_control_node-1] [INFO] [1709406599.843083413] [RRBotSensorPositionFeedback]: Reading...
[ros2_control_node-1] [INFO] [1709406599.843086367] [RRBotSensorPositionFeedback]: Got measured velocity nan
[ros2_control_node-1] [INFO] [1709406599.843089180] [RRBotSensorPositionFeedback]: Got state 0.00000 for joint 'joint2'!
[ros2_control_node-1] [INFO] [1709406599.843092310] [RRBotSensorPositionFeedback]: Joints successfully read!
[... cut ...]
This demo example is an interesting special case since the `on_activate()` artificially takes a couple of seconds which is usually probably not the case.

Or is it a requirement that `on_activate()` is required to be real-time-safe and run in one control cycle? Now that I'm aware of the reason I remember I've seen something similar on the UR driver before we restructured things in `on_configure` and `on_activate` correctly.

So my main question would be: Is this behavior to be expected and the requirements for the `on_activate()` aren't fulfilled or do we have to update something in the controller_manager / resource_manager?

christophfroehlich, 2024-03-02:
Thanks for analyzing this.

> This demo example is an interesting special case since the `on_activate()` artificially takes a couple of seconds which is usually probably not the case.
>
> Or is it a requirement that `on_activate()` [...]

I just checked the [docs](https://control.ros.org/master/doc/differences_to_ros1/differences_to_ros1.html#controller-migration) about that. It seems that this should be real-time safe:

        on_configure() - reads parameters and configures controller.

        on_activate() - called when controller is activated (started) (real-time)

        on_deactivate() - called when controller is deactivated (stopped) (real-time)

We should check if this is really the case in the CM (@saikishor maybe?), clarify that in [the docs here](https://control.ros.org/master/doc/ros2_control/hardware_interface/doc/writing_new_hardware_component.html) and fix the examples of the demos.

> Since urdf loading and activating the hardware isn't blocking the `update()` loop and the resource manager calls `write()` for all command interface independent of their state.
> 
> So, one way of "fixing" this short term would be a check in the `write()` method whether the command is `NaN` and long term to finally implement command interfaces that are not available for writing when not being active as discussed in [ros-controls/ros2_control#884](https://github.com/ros-controls/ros2_control/pull/884)

I'm fine with the fix, and agree for the need for the "final" solution upstream.

fmauch, 2024-03-02:
Short term solution now included in #456

saikishor, 2024-03-02:
Hello! 

I'd rather go with a proper fix, because if this happened in the demos, this might happen in reality, we would need to check why it is happening and find a proper solution for upstream. What do you guys think?

fmauch, 2024-03-02:
Yes, I think a proper fix would be better. That's why I didn't mark this issue to be resolved by #456. I just like this issue to not block migrating towards the topic inside the demos.

saikishor, 2024-03-02:
After looking at the code, I believe this is happening because we call the `write` method also for the `inactive` components 
https://github.com/ros-controls/ros2_control/blob/master/hardware_interface%2Fsrc%2Fsystem.cpp#L247 and https://github.com/ros-controls/ros2_control/blob/master/hardware_interface%2Fsrc%2Factuator.cpp#L251

The only fix I can think of is how @fmauch has handled it in the hardware itself. I'm wondering why this is not happening with the parameter. I would like to check this part.

If we need to properly fix it, I think we might need a new method called `dynamic_configure` where all the `gpio` interfaces go in and then the main hardware command interfaces can go in the `write` method, in this way we can call this new `dynamic_configure` method and avoid calling `write` method in the `inactive` state.

christophfroehlich, 2024-07-18:
@saikishor should we open an issue in ros2_control and close this one here?

saikishor, 2024-07-18:
> @saikishor should we open an issue in ros2_control and close this one here?

Sure, let's open an issue in ros2_control and leave this open still. What do you think?

christophfroehlich, 2024-07-20:
Demos fixed with #456 -> I created a new issue in ros2_control repo.

closed 2024-07-20 by christophfroehlich

christophfroehlich, 2024-08-02:
@fmauch @saikishor should we remove the `std::is_finite` check now as the original problem is solved (see #1632)?

saikishor, 2024-08-02:
> @fmauch @saikishor should we remove the `std::is_finite` check now as the original problem is solved (see #1632)?

I would say yes. this way we can be aware if this breaks again