ros-controls/ros2_control_demos #477: Can not start 2 controller simultaly for mobil robot with arm

FrGe2016, 2024-03-22:
I have a mobile robot with a small arm. When try to start the diff drive for the base with the joint_trajectory_controller for the arm . The first controller is loaded then fails living the second controller live.Individualy both controller are working properly.

Her is the controller.yaml followed by the runtime error. 

My goal is to have simultaneous motions of the of the base and the arm.

controller_manager:
  ros__parameters:                            # 'est ce fichier qui est utilisé dans d2.real
    update_rate: 20  # Hz

    base_controller:
      type: diff_drive_controller/DiffDriveController

    arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

      
    joint_state_broadcaster:                                   
      type: joint_state_broadcaster/JointStateBroadcaster      

      

arm_controller:
  ros__parameters:
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    allow_partial_joints_goal: true
    joints:
      - elbow
      - wrist
      - rotator
      - lateral
      - gripper
      - gripper_top

base_controller:
  ros__parameters:
    left_wheel_names: ["left_wheel_joint"]
    right_wheel_names: ["right_wheel_joint"]

    wheel_separation: 0.8
    #wheels_per_side: 1  # actually 2, but both are controlled by 1 signal
    wheel_radius: 0.25

    wheel_separation_multiplier: 1.0
    left_wheel_radius_multiplier: 1.0
    right_wheel_radius_multiplier: 1.0

    publish_rate: 50.0
    odom_frame_id: odom_dif
    base_frame_id: base_footprint2
    pose_covariance_diagonal : [0.001, 0.001, 0.001, 0.001, 0.001, 0.01]
    twist_covariance_diagonal: [0.001, 0.001, 0.001, 0.001, 0.001, 0.01]

    open_loop: true
    enable_odom_tf: true

    cmd_vel_timeout: 0.5
    #publish_limited_velocity: true
    #velocity_rolling_window_size: 10

    # Velocity and acceleration limits
    # Whenever a min_* is unspecified, default to -max_*
    linear.x.has_velocity_limits: true
    linear.x.has_acceleration_limits: true
    linear.x.has_jerk_limits: false
    linear.x.max_velocity: 0.3
    linear.x.min_velocity: -0.3
    linear.x.max_acceleration: 1.0
    linear.x.max_jerk: 0.0
    linear.x.min_jerk: 0.0

    angular.z.has_velocity_limits: true
    angular.z.has_acceleration_limits: true
    angular.z.has_jerk_limits: false
    angular.z.max_velocity: 0.3
    angular.z.min_velocity: -0.3
    angular.z.max_acceleration: 1.0
    angular.z.min_acceleration: -1.0
    angular.z.max_jerk: 0.0
    angular.z.min_jerk: 0.0

Run time error :+[spawner-4] [INFO] [1711075842.059151005] [spawner_arm_controller]: Loaded arm_controller
[ros2_control_node-1] [INFO] [1711075842.144777239] [controller_manager]: Configuring controller 'arm_controller'
[ros2_control_node-1] [INFO] [1711075842.145505820] [arm_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-1] [INFO] [1711075842.145875860] [arm_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ros2_control_node-1] [INFO] [1711075842.145925492] [arm_controller]: Using 'splines' interpolation method.
[ros2_control_node-1] [INFO] [1711075842.158437313] [arm_controller]: Controller state will be published at 50.00 Hz.
[ros2_control_node-1] [INFO] [1711075842.167363825] [arm_controller]: Goals with partial set of joints are allowed
[ros2_control_node-1] [INFO] [1711075842.167431143] [arm_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-4] [INFO] [1711075842.300252579] [spawner_arm_controller]: Configured and activated arm_controller
[INFO] [spawner-4]: process has finished cleanly [pid 883506]
[ros2_control_node-1] [INFO] [1711075842.651619667] [controller_manager]: Loading controller 'base_controller'
[spawner-5] [INFO] [1711075842.801189537] [spawner_base_controller]: Loaded base_controller
[ros2_control_node-1] [INFO] [1711075842.888846258] [controller_manager]: Configuring controller 'base_controller'
[ros2_control_node-1] [ERROR] [1711075842.958027289] [resource_manager]: Not acceptable command interfaces combination: 
[ros2_control_node-1] Start interfaces: 
[ros2_control_node-1] [
[ros2_control_node-1]   left_wheel_joint/velocity
[ros2_control_node-1]   right_wheel_joint/velocity
[ros2_control_node-1] ]
[ros2_control_node-1] Stop interfaces: 
[ros2_control_node-1] [
[ros2_control_node-1] ]
[ros2_control_node-1] Not existing: 
[ros2_control_node-1] [
[ros2_control_node-1]  left_wheel_joint/velocity
[ros2_control_node-1]  right_wheel_joint/velocity
[ros2_control_node-1] ]
[ros2_control_node-1] 
[ros2_control_node-1] [ERROR] [1711075842.958119589] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
[spawner-5] [ERROR] [1711075842.963628502] [spawner_base_controller]: Failed to activate controller

christophfroehlich, 2024-03-22:
I don't see why it should not work now. How do you spawn your controllers? Can you please add the output of `ros2 control list_hardware_interfaces ` and `ros2 control list_hardware_components -v` `ros2 control list_controllers -v`?

christophfroehlich, 2024-05-07:
Closing due to inactivity.

closed 2024-05-07 by christophfroehlich