ros-controls/ros2_control_demos #490: Implementation Query - Example 14 (Actuator and Sensor Interface)?

ImTheMan153, 2024-04-07:
Hello,
I have been working with Ros2 control for a very long time and it is wonderful. Now, I am trying to implement a z axis with stepper control. So I just need a trigger signal that sends true (int 1) to my arduino based on my system state from ros ecosystem and after the task is done my Microcontroller would send back true in another variable through serial. I have a custom serial interface already implemented so all I need to do is call two different functions from my comms file which is void sendRequest(triggersignal) and void readState(zaxisstate) and I update another feedback topic based (provided by ros2 control) on the information I get. I thought I could use actuator interface although read function wasn't there in the example, I tried adding it. It doesn't really work. Also when my stepper does the task it gets stuck in a while loop as the steps are passed to the driver. Does it block my serial interface? In this case, which example I should be looking for and what is the best method to implement this?
Thank you.

christophfroehlich, 2024-04-07:
You can see the difference of `system_interface`, `actuator_interface`, and `sensor_interface` [here](https://control.ros.org/master/doc/ros2_control/hardware_interface/doc/hardware_interface_types_userdoc.html#examples). 
You could try to implement both inside a system_interface like in [example_4](https://control.ros.org/master/doc/ros2_control_demos/example_4/doc/userdoc.html) or separate it into sensor and actuator as done in [example_14](https://control.ros.org/master/doc/ros2_control_demos/example_14/doc/userdoc.html).

christophfroehlich, 2024-05-07:
Closing due to inactivity.

closed 2024-05-07 by christophfroehlich