ros-controls/ros2_control_demos #493: Waiting for '/controller_manager' services to be available

FeetPro, 2024-04-09:
Hello,
I've followed the demo for r6bot 
I try to launch : ros2 launch ros2_control_demo_example_7 r6bot_controller.launch.py but it appears some problems. 

[INFO] [launch]: All log files can be found below /home/feet/.ros/log/2024-04-10-00-50-38-159675-feet-VirtualBox-7201
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [7203]
[INFO] [robot_state_publisher-2]: process started with pid [7205]
[INFO] [spawner-3]: process started with pid [7207]
[robot_state_publisher-2] [INFO] [1712685038.513462644] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1712685038.513562991] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-2] [INFO] [1712685038.513572347] [robot_state_publisher]: got segment link_1
[robot_state_publisher-2] [INFO] [1712685038.513578017] [robot_state_publisher]: got segment link_2
[robot_state_publisher-2] [INFO] [1712685038.513582973] [robot_state_publisher]: got segment link_3
[robot_state_publisher-2] [INFO] [1712685038.513587776] [robot_state_publisher]: got segment link_4
[robot_state_publisher-2] [INFO] [1712685038.513592705] [robot_state_publisher]: got segment link_5
[robot_state_publisher-2] [INFO] [1712685038.513597537] [robot_state_publisher]: got segment link_6
[robot_state_publisher-2] [INFO] [1712685038.513602417] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1712685038.513607275] [robot_state_publisher]: got segment world
[ros2_control_node-1] [INFO] [1712685038.556823162] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-1] [INFO] [1712685038.864750383] [controller_manager]: update rate is 10 Hz
[ros2_control_node-1] [INFO] [1712685038.867007474] [controller_manager]: RT kernel is recommended for better performance
[spawner-3] [INFO] [1712685041.164756425] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-3] [INFO] [1712685043.253130650] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-3] [INFO] [1712685045.349502856] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-3] [INFO] [1712685047.446210621] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-3] [ERROR] [1712685049.102395644] [spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner-3]: process has died [pid 7207, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --ros-args'].
[INFO] [spawner-4]: process started with pid [7238]
[INFO] [rviz2-5]: process started with pid [7240]
[rviz2-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[rviz2-5] [INFO] [1712685050.815188238] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1712685050.817931829] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-5] [INFO] [1712685051.015317686] [rviz2]: Stereo is NOT SUPPORTED
[spawner-4] [INFO] [1712685052.743597759] [spawner_r6bot_controller]: Waiting for '/controller_manager' services to be available
[spawner-4] [INFO] [1712685054.806592732] [spawner_r6bot_controller]: Waiting for '/controller_manager' services to be available
[spawner-4] [INFO] [1712685056.971315333] [spawner_r6bot_controller]: Waiting for '/controller_manager' services to be available
[INFO] [rviz2-5]: process has finished cleanly [pid 7240]
[spawner-4] [INFO] [1712685059.073277045] [spawner_r6bot_controller]: Waiting for '/controller_manager' services to be available
[spawner-4] [ERROR] [1712685060.655098531] [spawner_r6bot_controller]: Controller manager not available
[ERROR] [spawner-4]: process has died [pid 7238, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner r6bot_controller -c /controller_manager --ros-args'].
Waiting for '/controller_manager' services to be available

I waste too much time and don't know how to fix it, i am a new beginner in Ros2.
i use ROS2 Humble and Ubuntu 22.04.

christophfroehlich, 2024-04-09:
Have you used the humble branch of this repository? I just tested it successfully.

FeetPro, 2024-04-10:
Do I have to install dependent packages in ros2_control_demos.humble.repos? And build it in workspace?
Sorry for my lack of knowledge.

christophfroehlich, 2024-04-10:
Can you please answer my question? And it is sufficient if you install the deps with rosdep as written in [our docs](https://control.ros.org/humble/doc/ros2_control_demos/doc/index.html#build-from-debian-packages).

FeetPro, 2024-04-10:
i find the problem when i see the docs you gave me, i haven't installed humble branch
thank you @christophfroehlich

closed 2024-04-10 by FeetPro