ros-controls/ros2_control_demos #564: carlikebot with teleop_twist_keyboard control

0jeetparmar0, 2024-08-07:
does anyone know how can i use teleop_twist_keyboard with carlikebot in ros2 humble 

as this video uses teleop_twist with diff_bot robot 

https://youtu.be/J02jEKawE5U?si=zKH39WVIPKfwr0t0

thanks and regards

christophfroehlich, 2024-08-07:
Have you had a look into the [readme](https://index.ros.org/p/teleop_twist_keyboard/github-ros2-teleop_twist_keyboard/#humble) of teleop_twist_keyboard?
You need to remap the topic, and take care that you use the same settings for the stamped version of the twist message: `use_stamped_vel` is default false on humble, but also the default for the teleop_twist_keyboard's `stamped` parameter.

0jeetparmar0, 2024-08-09:
> Have you had a look into the [readme](https://index.ros.org/p/teleop_twist_keyboard/github-ros2-teleop_twist_keyboard/#humble) of teleop_twist_keyboard? You need to remap the topic, and take care [...]

thanks for response but i am new to ros and i can not understand how do i do it .....

christophfroehlich, 2024-08-12:
The remap command is written in the first section of the readme, just use the topic necessary for the bicycle_steering_controller

`ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args --remap cmd_vel:=/bicycle_steering_controller/reference`

If you have further questions, I suggest that you open a question on https://robotics.stackexchange.com/questions, because this topic is not related to our repository here.

closed 2024-08-12 by christophfroehlich