ros-controls/ros2_control_demos #568: PID controller and diff drive controller chaining

jclinton830, 2024-08-16:
**Is your feature request related to a problem? Please describe.**

Is it worth making a demo on how to chain the diff drive controller with the pid controller given that a lot of wheeled robot users would want to do this in their setup? 

**Describe the solution you'd like**
A demo on how to do this would be very beneficial to many people, so not much time is wasted on trial and error, which can be easily applied to other controllers such as Ackermann and steering controllers.

christophfroehlich, 2024-08-16:
Hi, yes I would find it useful. If you want to contribute that, copy example_2, rename it for example_16 (the next free number) and apply your changes.

christophfroehlich, 2025-03-13:
Implemented with #710

closed 2025-03-13 by christophfroehlich