ros-controls/ros2_control_demos #605: example_15: Wrong ctrl config

christophfroehlich, 2024-10-27:
https://github.com/ros-controls/ros2_control_demos/actions/runs/11536820407
```
Error: ROR] [spawner-9]: process has died [pid 22202, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner position_trajectory_controller --inactive --param-file /home/runner/work/ros2_control_demos/ros2_control_demos/.work/target_ws/install/ros2_control_demo_example_15/share/ros2_control_demo_example_15/config/multi_controller_manager_rrbot_generic_controllers.yaml --ros-args -r __ns:=/rrbot_1'].
Error: s2_control_node-1] [ERROR] [1729998296.023592492] [rrbot_1.controller_manager]: The 'type' param was not defined for 'forward_position_controller'.
```
@saikishor could you have a look pls?

saikishor, 2024-10-27:
> https://github.com/ros-controls/ros2_control_demos/actions/runs/11536820407
> ```
> Error: ROR] [spawner-9]: process has died [pid 22202, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner [...]

Sure!

christophfroehlich, 2024-10-27:
Ah, these are the binary builds -> semi-binary builds are still fine -> will solve itself after merging the releases into rosdistro

closed 2024-10-27 by christophfroehlich