ros-controls/ros2_control_demos #662: Accodhttps://control.ros.org/humble/doc/ros2_control_demos/doc/index.html#examples

WTTien, 2024-12-12:
https://control.ros.org/humble/doc/ros2_control_demos/doc/index.html#examples > "Build from source" section.

ROS DISTRO: Humble

Steps failed at colcon build --symlink-install, reason being ros2_control version 4.0.0 higher is required. However for ROS Distro Humble, it seems like ros2_control packages latest release is 2.45.0, which make the build fails?

christophfroehlich, 2024-12-12:
This comes from the following line on the `master` branch
https://github.com/ros-controls/ros2_control_demos/blob/master/example_1/CMakeLists.txt#L20

be aware that you either have to switch to the appropriate branch here (which is `humble`), or you have to build the whole ros2_control stack from source (called **Development Version** in the docs)

closed 2024-12-12 by christophfroehlich

WTTien, 2024-12-13:
I am able to make it work now. Not too sure which steps I missed on the first time round but thank you very much for the help!