ros-controls/ros2_control_demos #685: Example 7: Remove empty callbacks in controllers and update docs

christophfroehlich, 2025-01-07:
## Background

Related to https://github.com/ros-controls/ros2_controllers/issues/1474

IMHO we could remove the empty callbacks
https://github.com/ros-controls/ros2_control_demos/blob/40c59eeac5475df0ab32a2e3a6e9d276a3ece827/example_7/controller/include/ros2_control_demo_example_7/r6bot_controller.hpp#L65-L72

or at least add a comment, that they are just for demonstration purposes and are optional.

At least we should update the docs, that the overrides of the lifecycle transitions are optional and not mandatory
https://github.com/ros-controls/ros2_control_demos/blob/40c59eeac5475df0ab32a2e3a6e9d276a3ece827/example_7/doc/userdoc.rst?plain=1#L336-L365

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control_demos/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/galactic/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control_demos`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control_demos` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control_demos repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/galactic/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

saikishor, 2025-01-07:
I agree 👍🏽

saikishor, 2025-01-15:
Hello @nakul-py!

Remember also this pending PR : https://github.com/ros-controls/ros2_controllers/pull/1475 :)

Thank you

AdityaPawar162, 2025-03-06:
Hello @saikishor @christophfroehlich,
I was wondering if I could contribute to this issue. Would that be possible?

christophfroehlich, 2025-03-06:
@nakul-py have you started working on this?

AdityaPawar162, 2025-03-07:
Hello @christophfroehlich @saikishor,
Can I raise a PR for this issue?

christophfroehlich, 2025-03-07:
yes, that's how it works :)

AdityaPawar162, 2025-03-07:
Hello @christophfroehlich,
I have submitted a PR [#731].Please let me know if it needs any improvement.Looking forward for the response.

Regards,
Aditya

closed 2025-03-07 by saikishor