ros-controls/ros2_control_demos #704: Incorrect command path listed in Humble Demo Installation guide

coffinfish, 2025-01-27:
**Describe the bug**
In the following [documentation](https://control.ros.org/humble/doc/ros2_control_demos/doc/index.html#to-run-the-ros2-control-demos), it seems like the command to view example_1 robot is wrong under "To View The Robot" > "Terminal 2"

When I follow the guide, the robot does not show when I run this command:
`rviz2 -d src/ros2_control_demos/example_1/description/rviz/rrbot.rviz`

It seems the correct path is:
`rviz2 -d src/ros2_control_demos/ros2_control_demo_description/rrbot/rviz/rrbot.rviz`

The same path appears in "To run the ros2_control demos[](https://control.ros.org/humble/doc/ros2_control_demos/doc/index.html#to-run-the-ros2-control-demos)"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [documentation](https://control.ros.org/humble/doc/ros2_control_demos/doc/index.html#installation)
2. Scroll down to To View the Robot > Terminal 2
4. Note the path of the rviz file. (`src/ros2_control_demos/example_1/description/rviz/rrbot.rviz`)
5. Go to github repository > Attempt to locate file > The rviz directory in example_1 does not exist, but the file exists in `ros2_control_demo_description/rrbot/` instead

**Expected behavior**
Robot should appear when running the command.

**Screenshots**
If applicable, add screenshots to help explain your problem.
Running `rviz2 -d src/ros2_control_demos/example_1/description/rviz/rrbot.rviz` leads to an empty RViz:
[image]

Running `rviz2 -d src/ros2_control_demos/ros2_control_demo_description/rrbot/rviz/rrbot.rviz` has proper result:
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 
 - Version: Humble, but I saw the same path in other documentation versions like Jazzy and Foxy, but the setup might be different for those versions?
- Nothing unusual

**Additional context**
No modifications, just following the guide. This is my first issue so I apologise if I did not do it correctly, please let me know if there's anything I can improve on.

christophfroehlich, 2025-01-27:
You are right, I forgot to update the general instructions. Do you want to fix that and open a PR?

coffinfish, 2025-01-27:
Did it here:
https://github.com/ros-controls/ros2_control_demos/pull/705

Apologies if I messed anything up. (I couldn't figure out how to test the changes I made even though it was only two lines)

PR #705 (2025-01-27): Fixed command path to rviz file.
Fixes #704 
Changed `rviz2 -d src/ros2_control_demos/example_1/description/rviz/rrbot.rviz` to 
`rviz2 -d src/ros2_control_demos/ros2_control_demo_description/rrbot/rviz/rrbot.rviz` where found.
PR #706 (2025-01-27): Fixed command path to rviz file. (backport #705)
Fixes #704 
Changed `rviz2 -d src/ros2_control_demos/example_1/description/rviz/rrbot.rviz` to 
`rviz2 -d src/ros2_control_demos/ros2_control_demo_description/rrbot/rviz/rrbot.rviz` where found.<hr>This is an automatic backport of pull request #705 done by [Mergify](https://mergify.com).
closed 2025-01-27 by christophfroehlich