ros-controls/ros2_control_demos #741: Can somebody help me with my controller_manager

RegenKraeker80, 2025-03-14:
Hi guys,

I am currently working on my Bachelor Thesis and have some issues regarding the Controller_manager on ROS2 Humble. I wrote a Hardware_interface which is called argo_drive_hardware.ccp an it is starting just fine when i run the ros2_control_node. But the controller_manager is not available for my joint_state_publisher and my tricycle_controller. I'm launching my Workspace using a Launch file Launch_ros2_ws.launch.py.
Did somebody have this issue before?
Here is the link to my Git Repository its a bit chaotic sorry it's my first time working with ROS.
https://github.com/RegenKraeker80/Argo_Drive_Robot_Workspace

Thank you all in advance

[image]

christophfroehlich, 2025-03-14:
Can you confirm that our demos work, e.g., [example_1](https://control.ros.org/humble/doc/ros2_control_demos/doc/index.html#examples)?

RegenKraeker80, 2025-03-15:
@christophfroehlich 
The examples work just fine. 
The controller_manager starts my plugin. But then stops working.

christophfroehlich, 2025-03-15:
If you want us to help you, please give us more info like what launch file you are starting etc. And please don't post screenshots of text.

RegenKraeker80, 2025-03-15:
Sure, 

Sorry it's my first time using ROS 
I'm launching my robot using the following command for my launch file Launch_ros2_ws.py:

sudo -E bash -c 'source /opt/ros/humble/setup.bash && source /home/adrian/ros2_ws/install/setup.bash && ros2 launch ros2_ws_robot_launch Launch_ros2_ws.py'

I'm using a custom hardware interface plugin called argo_drive_hardware.cpp. The controller_manager starts the plugin successfully, and the plugin properly connects to my EtherCAT slave. The read() and write() functions are called without issues as well.

However, after that, the controller_manager is no longer available. I verified this by trying to list the controllers or query the controller_manager node, but it seems to be missing or i get the warning:

waiting for service /controller_manager/... to become available

I have enabled debugging, but no errors appear in the output.

saikishor, 2025-03-15:
Hello @RegenKraeker80 

The CM stopping to respond is indeed very strange. Can you try not to spawn any controller for a few min after starting your hardware and see if the services are still available or not?

Can you post the version tags of the ros2_control packages and ros2_controllers?

RegenKraeker80, 2025-03-15:
Hello @saikishor 

My versions are: 
ros-humble-ros2-control-test-assets/jammy,now 2.48.0-1jammy.20250207.195608 amd64 [installed]
ros-humble-ros2-control/jammy,now 2.48.0-1jammy.20250207.202533 amd64 [installed]
ros-humble-ros2-controllers/jammy,now 2.41.0-1jammy.20250207.204752 amd64 [installed]
ros-humble-ros2-controllers/jammy,now 2.41.0-1jammy.20250207.204752 amd64 [installed]

I did push everything out of my hardware interface and just return SUCCESS every time to check if it was a problem with my EtherCAT Master but that didn't  do anything. 

The controller_manager won't even start up at all and there are no Error's comming up. And the controller_manager node is not available. 
the only node thats starting up ist /ros2_control and the services that are available are like /ros2_control/controller_manager/...

Thank you in advance for helping me, I've been struggling with that one for two weeks.

saikishor, 2025-03-15:
> The controller_manager won't even start up at all and there are no Error's comming up. And the controller_manager node is not available. 
> the only node thats starting up ist /ros2_control and the services that are available are like /ros2_control/controller_manager/...

I see that it is starting properly. You just namespaced the ros2_control_node. If the services are available under the ros2_control namespace. You just have to remove the namespace when you launch the ros2_control_node.

saikishor, 2025-03-18:
@RegenKraeker80 did you make it work?

RegenKraeker80, 2025-03-19:
@saikishor I made it work 
Thank you so much. And I'm sorry for my late answer and wasting your time.

closed 2025-03-19 by christophfroehlich