ros-controls/ros2_control_demos #765: Segfault in example_13 tests during deactivation (jazzy)

christophfroehlich, 2025-03-31:
```
     [INFO] [publisher_forward_position_controller-9]: sending signal 'SIGINT' to process[publisher_forward_position_controller-9]
    [INFO] [publisher_forward_position_controller-8]: sending signal 'SIGINT' to process[publisher_forward_position_controller-8]
    [INFO] [publisher_forward_position_controller-7]: sending signal 'SIGINT' to process[publisher_forward_position_controller-7]
    [INFO] [robot_state_publisher-2]: sending signal 'SIGINT' to process[robot_state_publisher-2]
    [INFO] [ros2_control_node-1]: sending signal 'SIGINT' to process[ros2_control_node-1]
    [robot_state_publisher-2] [INFO] [1743389866.479201754] [rclcpp]: signal_handler(signum=2)
    [ros2_control_node-1] [INFO] [1743389866.479254633] [rclcpp]: signal_handler(signum=2)
    [ros2_control_node-1] [INFO] [1743389866.479357045] [controller_manager]: Shutdown request received....
    [ros2_control_node-1] [INFO] [1743389866.479417678] [controller_manager]: Shutting down all controllers in the controller manager.
    [ros2_control_node-1] [INFO] [1743389866.479479163] [controller_manager]: Deactivating controller 'rrbot_with_sensor_joint_state_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.479534387] [controller_manager]: Shutting down controller 'rrbot_with_sensor_joint_state_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.479769667] [controller_manager]: Deactivating controller 'threedofbot_pid_gain_controller'
    [ros2_control_node-1] [INFO] [1743389866.479829719] [controller_manager]: Shutting down controller 'threedofbot_pid_gain_controller'
    [ros2_control_node-1] [INFO] [1743389866.479901744] [controller_manager]: Deactivating controller 'rrbot_external_fts_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.479943162] [controller_manager]: Shutting down controller 'rrbot_external_fts_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.480001390] [controller_manager]: Shutting down controller 'rrbot_joint_state_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.480056794] [controller_manager]: Deactivating controller 'joint_state_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.480090838] [controller_manager]: Shutting down controller 'joint_state_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.480136483] [controller_manager]: Deactivating controller 'rrbot_with_sensor_fts_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.480155088] [controller_manager]: Shutting down controller 'rrbot_with_sensor_fts_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.480222424] [controller_manager]: Shutting down controller 'rrbot_position_controller'
    [ros2_control_node-1] [INFO] [1743389866.480272498] [controller_manager]: Deactivating controller 'threedofbot_position_controller'
    [ros2_control_node-1] [INFO] [1743389866.480318674] [controller_manager]: Shutting down controller 'threedofbot_position_controller'
    [ros2_control_node-1] [INFO] [1743389866.480360402] [controller_manager]: Deactivating controller 'threedofbot_joint_state_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.480390378] [controller_manager]: Shutting down controller 'threedofbot_joint_state_broadcaster'
    [ros2_control_node-1] [INFO] [1743389866.480423992] [controller_manager]: Deactivating controller 'rrbot_with_sensor_position_controller'
    [ros2_control_node-1] [INFO] [1743389866.480455821] [controller_manager]: Shutting down controller 'rrbot_with_sensor_position_controller'
    [ros2_control_node-1] [INFO] [1743389866.480523237] [resource_manager]: 'deactivate' hardware 'FakeThreeDofBot' 
    [ros2_control_node-1] [INFO] [1743389866.480542563] [resource_manager]: Successful 'deactivate' of hardware 'FakeThreeDofBot'
    [ros2_control_node-1] [INFO] [1743389866.480550669] [resource_manager]: 'shutdown' hardware 'FakeThreeDofBot' 
    [ros2_control_node-1] Stack trace (most recent call last) in thread [INFO] [1743389866.480556900] [resource_manager]: Successful 'shutdown' of hardware 'FakeThreeDofBot'
    [ros2_control_node-1] [INFO] [1743389866.480685371] [resource_manager]: 'deactivate' hardware 'RRBotSystemWithSensor' 
    [ros2_control_node-1] [INFO] [1743389866.480693686] [controller_manager.hardware_component.system.RRBotSystemWithSensor]: Deactivating ...please wait...
    [ros2_control_node-1] 48189:
    [ros2_control_node-1] #18   Object "", at 0xffffffffffffffff, in 
    [ros2_control_node-1] #17   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fab3372aa33, in __clone
    [ros2_control_node-1] #16   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fab3369daa3, in 
    [ros2_control_node-1] #15   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x7fab33a16db3, in 
    [ros2_control_node-1] #14   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x5597f3cbb907, in std::thread::_State_impl<std::thread::_Invoker<std::tuple<main::{lambda()#1}> > >::_M_run()
    [ros2_control_node-1] #13   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x5597f3cbb923, in std::thread::_Invoker<std::tuple<main::{lambda()#1}> >::operator()()
    [ros2_control_node-1] #12   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x5597f3cbb94f, in void std::thread::_Invoker<std::tuple<main::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>)
    [ros2_control_node-1] #11   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x5597f3cbb9a2, in std::__invoke_result<main::{lambda()#1}>::type std::__invoke<main::{lambda()#1}>(main::{lambda()#1}&&)
    [ros2_control_node-1] #10   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x5597f3cbb9df, in void std::__invoke_impl<void, main::{lambda()#1}>(std::__invoke_other, main::{lambda()#1}&&)
    [ros2_control_node-1] #9    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x5597f3cb952b, in main::{lambda()#1}::operator()() const
    [ros2_control_node-1] #8    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/libcontroller_manager.so", at 0x7fab344873fe, in controller_manager::ControllerManager::update(rclcpp::Time const&, rclcpp::Duration const&)
    [ros2_control_node-1] #7    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_interface/lib/libcontroller_interface.so", at 0x7fab33515362, in controller_interface::ControllerInterfaceBase::trigger_update(rclcpp::Time const&, rclcpp::Duration const&)
    [ros2_control_node-1] #6    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/joint_state_broadcaster/lib/libjoint_state_broadcaster.so", at 0x7fab29568d9b, in joint_state_broadcaster::JointStateBroadcaster::update(rclcpp::Time const&, rclcpp::Duration const&)
    [ros2_control_node-1] #5    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/joint_state_broadcaster/lib/libjoint_state_broadcaster.so", at 0x7fab29577f48, in std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, double> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> [... cut ...]
Error: ROR] [ros2_control_node-1]: process has died [pid 48120, exit code -11, cmd '/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node --ros-args --params-file /home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/ros2_control_demo_example_13/share/ros2_control_demo_example_13/config/three_robots_controllers.yaml'].
    test_exit_codes (ros2_control_demo_example_13.TestDescriptionCraneShutdown.test_exit_codes)
    Check if the processes exited normally. ... FAIL
```

christophfroehlich, 2025-04-11:
Also happens on humble
```
2025-04-09T03:08:22.6166955Z     [ros2_control_node-1] [INFO] [1744168032.847932927] [controller_manager.resource_manager.hardware_component.system.RRBotSystemWithSensor]: Deactivating ...please wait...
2025-04-09T03:08:22.6167497Z     [ros2_control_node-1] Stack trace (most recent call last) in thread 40025:
2025-04-09T03:08:22.6167763Z     [ros2_control_node-1] #17   Object "", at 0xffffffffffffffff, in 
2025-04-09T03:08:22.6168168Z     [ros2_control_node-1] #16   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f93a9a8ba03, in __clone
2025-04-09T03:08:22.6168641Z     [ros2_control_node-1] #15   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f93a99faac2, in 
2025-04-09T03:08:22.6169045Z     [ros2_control_node-1] #14   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f93a9c8b252, in 
2025-04-09T03:08:22.6170213Z     [ros2_control_node-1] #13   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x55d84f820bbb, in std::thread::_State_impl<std::thread::_Invoker<std::tuple<main::{lambda()#1}> > >::_M_run()
2025-04-09T03:08:22.6171609Z     [ros2_control_node-1] #12   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x55d84f820bd7, in std::thread::_Invoker<std::tuple<main::{lambda()#1}> >::operator()()
2025-04-09T03:08:22.6172980Z     [ros2_control_node-1] #11   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x55d84f820c03, in void std::thread::_Invoker<std::tuple<main::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>)
2025-04-09T03:08:22.6174327Z     [ros2_control_node-1] #10   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x55d84f820c56, in std::__invoke_result<main::{lambda()#1}>::type std::__invoke<main::{lambda()#1}>(main::{lambda()#1}&&)
2025-04-09T03:08:22.6175752Z     [ros2_control_node-1] #9    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x55d84f820c93, in void std::__invoke_impl<void, main::{lambda()#1}>(std::__invoke_other, main::{lambda()#1}&&)
2025-04-09T03:08:22.6176786Z     [ros2_control_node-1] #8    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/controller_manager/ros2_control_node", at 0x55d84f81f489, in main::{lambda()#1}::operator()() const
2025-04-09T03:08:22.6177616Z     [ros2_control_node-1] #7    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/controller_manager/lib/libcontroller_manager.so", at 0x7f93aa694f98, in controller_manager::ControllerManager::update(rclcpp::Time const&, rclcpp::Duration const&)
2025-04-09T03:08:22.6178588Z     [ros2_control_node-1] #6    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/joint_state_broadcaster/lib/libjoint_state_broadcaster.so", at 0x7f93a237b789, in joint_state_broadcaster::JointStateBroadcaster::update(rclcpp::Time const&, rclcpp::Duration const&)
2025-04-09T03:08:22.6183824Z     [ros2_control_node-1] #5    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/joint_state_broadcaster/lib/libjoint_state_broadcaster.so", at 0x7f93a2389bfc, in std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, double> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, double> > > > > >::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
2025-04-09T03:08:22.6188826Z     [ros2_control_node-1] #4    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/joint_state_broadcaster/lib/libjoint_state_broadcaster.so", at 0x7f93a238e183, in std::__detail::_Map_base<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double, std::hash<std::__cxx11::basic_string<char, std::cha [... cut ...] cator<char> >, double, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, double> > > >, true> const&, unsigned long) const
2025-04-09T03:08:22.6223777Z     [ros2_control_node-1] Segmentation fault (Address not mapped to object [0x7f948137c8d4])
```

christophfroehlich, 2025-04-17:
https://github.com/ros-controls/ros2_control_ci/actions/runs/14436292889/job/40477863471
@saikishor it seems that there is a conceptional issue, with a segfault of set_value during on_deactivate?
```
     [ros2_control_node-1] #10   Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/steering_controllers_library/lib/libsteering_controllers_library.so", at 0x7fef5473728f, in steering_controllers_library::SteeringControllersLibrary::on_deactivate(rclcpp_lifecycle::State const&)
    [ros2_control_node-1] #9    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/steering_controllers_library/lib/libsteering_controllers_library.so", at 0x7fef5475431f, in bool hardware_interface::LoanedCommandInterface::set_value<double>(double const&, unsigned int)
    [ros2_control_node-1] #8    Object "/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/install/steering_controllers_library/lib/libsteering_controllers_library.so", at 0x7fef5475c56e, in bool hardware_interface::CommandInterface::set_limited_value<double>(double const&)
```

The other issues above were with the JSB.

saikishor, 2025-04-17:
> https://github.com/ros-controls/ros2_control_ci/actions/runs/14436292889/job/40477863471 [@saikishor](https://github.com/saikishor) it seems that there is a conceptional issue, with a segfault of set_value during on_deactivate?

@christophfroehlich I couldn't reproduce this with the current `master` of ros2_control right now, so let's wait for it to happen again :)

christophfroehlich, 2025-10-20:
https://github.com/ros-controls/ros2_control_demos/issues/943

christophfroehlich, 2026-09-21:
https://github.com/ros-controls/ros2_control_demos/issues/1206