ros-controls/ros2_control_demos #772: I'm always frustrated when I have to duplicate controller.yaml files just to accommodate running a controller node in another namespace.

Smith-JackSmith, 2025-04-05:
Since the namespace is the only thing changing, the controller.yaml shouldn't have to change and should be valid for any namespace.

The controller should read the controller.yaml file independent of which namespace it's running in.

Only if a namespace implies a different configuration should another controller.yaml file be specified in the launch file.

I should not have to enter the namespace within the controller.yaml.  this is duplication of effort and induces error.

christophfroehlich, 2025-04-05:
This is a ROS 2 parameter issue. Have you tried to add a wildcard in front of your controller name?
https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Launch/Using-ROS2-Launch-For-Large-Projects.html#using-wildcards-in-yaml-files
we also use that in this repo
https://github.com/ros-controls/ros2_control_demos/blob/67756e85dbde8072236a5f39e3b148b0f346cbc3/example_15/bringup/config/multi_controller_manager_rrbot_generic_controllers.yaml#L9

Smith-JackSmith, 2025-04-06:
Ahh, I finally understand what this tutorial is attempting to communicate. 

Wouldn't it be nice if there was actual documentation about this instead of burying vital information in tutorials. 

Anyway, thanks for the information, solved my issue.

saikishor, 2025-04-06:
Heyy @Smith-JackSmith 

We have it documented here : https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#helper-scripts

Smith-JackSmith, 2025-04-06:
Actually, it's not documented well. And, I'm not complaining. 

Proper documentation would involve some kind of BNF (perhaps), or at least a complete detailing of controllers.yaml file allowed /mandatory elements and their purpose.

I've searched extensively and found no such documentation.

saikishor, 2025-04-06:
> Actually, it's not documented well. And, I'm not complaining. 
> 
> Proper documentation would involve some kind of BNF (perhaps), or at least a complete detailing of controllers.yaml file allowed /mandatory elements and their purpose.
> 
> I've searched extensively and found no such documentation. 

Sure! Can you contribute to this part? You can open a pull request with the documentation, so it would be helpful for the next person. It would be great if you can do that.

Smith-JackSmith, 2025-04-06:
> Sure! Can you contribute to this part? You can open a pull request with the documentation, so it would be helpful for the next person. It would be great if you can do that.

I can't understand how it's supposed to work because it's inadequately documented, so I can't contribute to make it better.

saikishor, 2025-04-06:
@Smith-JackSmith I just saw your example. Please understand that this is the limitation of wildcard entries in the ROS 2. If you have something to mention, then open an issue on rclcpp packages.

christophfroehlich, 2025-04-06:
Closing, as this is not an issue of ros2_control (at least not of the demos repository), but of rclcpp.
If you find that something in the docs of ros2_control is missing, please contribute your improvement!

closed 2025-04-06 by christophfroehlich