ros-controls/ros2_control_demos #864: Example_13 test is broken

christophfroehlich, 2025-07-29:
It seems that we broke example_13 (or at least the test) with https://github.com/ros-controls/ros2_control/pull/2347

```
     [ros2_control_node-1] [INFO] [1753760884.651091427] [controller_manager]: Activating controllers: [ threedofbot_joint_state_broadcaster threedofbot_pid_gain_controller ]
    [ros2_control_node-1] [WARN] [1753760884.651350703] [controller_manager]: Component 'FakeThreeDofBot' is in INACTIVE state, but has start interfaces to switch: 
    [NON-XML-CHAR-0x1B][91mCould not switch controllers since prepare command mode switch was rejected.[NON-XML-CHAR-0x1B][0m
    [ros2_control_node-1] Start interfaces: 
    [ros2_control_node-1] [
    [ros2_control_node-1]   threedofbot_joint1/pid_gain
    [ros2_control_node-1]   threedofbot_joint2/pid_gain
    [ros2_control_node-1]   threedofbot_joint3/pid_gain
    [ros2_control_node-1] ]
    [ros2_control_node-1] Stop interfaces: 
    [ros2_control_node-1] [
    [ros2_control_node-1] ]
    [ros2_control_node-1] 
Error: s2_control_node-1] [ERROR] [1753760884.651373715] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
```

https://github.com/ros-controls/ros2_control_demos/issues/859
https://github.com/ros-controls/ros2_control_demos/issues/861
https://github.com/ros-controls/ros2_control_demos/issues/862
https://github.com/ros-controls/ros2_control_demos/issues/863
https://github.com/ros-controls/ros2_control_ci/issues/421
https://github.com/ros-controls/ros2_control_ci/issues/422
https://github.com/ros-controls/ros2_control_ci/issues/423
https://github.com/ros-controls/ros2_control_ci/issues/424

christophfroehlich, 2025-07-31:
@saikishor I analyzed this failure. The demo starts the threedofbot_pid_gain_controller while the fake bot (generic system) is in inactive state.

> Configure ``FakeThreeDofBot`` and its joint state broadcaster and non-movement command interfaces. Call
> 
>   .. code-block:: shell
> 
>     ros2 control set_hardware_component_state FakeThreeDofBot inactive
>     ros2 control switch_controllers --activate threedofbot_joint_state_broadcaster threedofbot_pid_gain_controller

Your PR now throws an error here, even before the prepare_command_mode_switch method of the hardware is called. 

Is it possible to have command interfaces (like PID gains in this example) if the hardware is inactive?

saikishor, 2025-07-31:
@christophfroehlich right now it doesn't let you activate any interfaces if it is in an inactive state. We will have to implement the logic of differentiating interfaces soon.

christophfroehlich, 2025-07-31:
neither GPIO command is allowed in inactive state of the HW?

christophfroehlich, 2025-07-31:
Isn't this then wrong now?
https://github.com/ros-controls/ros2_control/blob/68e190ac5a4447150a7ea0a2457d945a8ab228d8/hardware_interface/include/hardware_interface/actuator_interface.hpp#L69-L74

PR #866 (2025-07-31): Don't activate controller with command_interfaces if hw is inactive
Fixes #864
closed 2025-07-31 by christophfroehlich