ros-controls/ros2_control_demos #865: Hardware components use old API

christophfroehlich, 2025-07-30:
## Background

All hardware components throw compile-time warnings like this one:
```
--- stderr: ros2_control_demo_example_4
/workspaces/ros2_rolling_ws/src/ros2_control_demos/example_4/hardware/rrbot_system_with_sensor.cpp: In member function ‘virtual hardware_interface::CallbackReturn ros2_control_demo_example_4::RRBotSystemWithSensorHardware::on_init(const hardware_interface::HardwareInfo&)’:
/workspaces/ros2_rolling_ws/src/ros2_control_demos/example_4/hardware/rrbot_system_with_sensor.cpp:38:49: warning: ‘virtual rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn hardware_interface::SystemInterface::on_init(const hardware_interface::HardwareInfo&)’ is deprecated: Use on_init(const HardwareComponentInterfaceParams & params) instead. [-Wdeprecated-declarations]
   38 |     hardware_interface::SystemInterface::on_init(info) !=
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
In file included from /workspaces/ros2_rolling_ws/src/ros2_control_demos/example_4/hardware/include/ros2_control_demo_example_4/rrbot_system_with_sensor.hpp:28,
                 from /workspaces/ros2_rolling_ws/src/ros2_control_demos/example_4/hardware/rrbot_system_with_sensor.cpp:19:
/workspaces/ros2_rolling_ws/src/ros2_control/hardware_interface/include/hardware_interface/system_interface.hpp:226:26: note: declared here
  226 |   virtual CallbackReturn on_init(const HardwareInfo & hardware_info)
      |                          ^~~~~~~
---
Finished <<< ros2_control_demo_example_4 [0.14s]
```

The API of the hardware components changed a bit, and got deprecated with 
ros-controls/ros2_control/issues/2344
For migration information, see also https://github.com/ros-controls/ros2_control/pull/2412

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

vvkvijaykrishna, 2025-08-11:
Hello @christophfroehlich, can I work on this?

christophfroehlich, 2025-09-28:
Fixed with #908

closed 2025-09-28 by christophfroehlich