ros-controls/ros2_control_demos #872: example_8 times out on CI from time to time

christophfroehlich, 2025-08-11:
**Describe the bug**
```
2025-08-11T04:00:39.5306635Z   1 package had stderr output: ros2_control_demo_example_8
2025-08-11T04:00:39.5306987Z   1 package had test failures: ros2_control_demo_example_8
2025-08-11T04:00:39.5307785Z ##[endgroup]
2025-08-11T04:00:39.5308138Z [32m'run_target_test' returned with code '0' after 20 min 45 sec[0m
2025-08-11T04:00:39.5308548Z 
2025-08-11T04:00:39.5309669Z [1m$ ( source /opt/ros/kilted/setup.bash && cd /home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws && colcon test-result --verbose; )[0m
2025-08-11T04:00:39.5311046Z build/ros2_control_demo_example_8/Testing/20250811-0356/Test.xml: 3 tests, 0 errors, 1 failure, 0 skipped
2025-08-11T04:00:39.5311796Z - run_example_8_launch
2025-08-11T04:00:39.5312130Z   <<< failure message
2025-08-11T04:00:39.5312365Z     -- run_test.py: extra environment variables:
2025-08-11T04:00:39.5312660Z      - PYTHONDONTWRITEBYTECODE=1
2025-08-11T04:00:39.5313238Z     -- run_test.py: invoking following command in '/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/build/ros2_control_demo_example_8':
2025-08-11T04:00:39.5315575Z      - /usr/bin/python3 -u -m pytest /home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/src/ros-controls/ros2_control_demos/example_8/test/test_rrbot_transmissions_system_position_only_launch.py -o cache_dir=/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/build/ros2_control_demo_example_8/ament_cmake_pytest/run_example_8_launch/.cache --junit-xml=/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/build/ros2_control_demo_example_8/test_results/ros2_control_demo_example_8/run_example_8_launch.xunit.xml --junit-prefix=ros2_control_demo_example_8
2025-08-11T04:00:39.5318167Z     ============================= test session starts ==============================
2025-08-11T04:00:39.5318834Z     platform linux -- Python 3.12.3, pytest-7.4.4, pluggy-1.4.0
2025-08-11T04:00:39.5319762Z     cachedir: build/ros2_control_demo_example_8/ament_cmake_pytest/run_example_8_launch/.cache
2025-08-11T04:00:39.5320628Z     rootdir: /home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws
2025-08-11T04:00:39.5322297Z     plugins: ament-pep257-0.19.2, ament-xmllint-0.19.2, ament-copyright-0.19.2, launch-testing-ros-0.28.2, ament-lint-0.19.2, launch-testing-3.8.2, ament-flake8-0.19.2, typeguard-4.1.5, cov-4.1.0, colcon-core-0.20.0, rerunfailures-12.0, repeat-0.9.3
2025-08-11T04:00:39.5323615Z     collected 1 item
2025-08-11T04:00:39.5323912Z   >>>
2025-08-11T04:00:39.5324812Z build/ros2_control_demo_example_8/test_results/ros2_control_demo_example_8/run_example_8_launch.xunit.xml: 1 test, 1 error, 0 failures, 0 skipped
2025-08-11T04:00:39.5325907Z - ros2_control_demo_example_8 run_example_8_launch.xunit.missing_result
2025-08-11T04:00:39.5326436Z   <<< error message
2025-08-11T04:00:39.5326742Z     The test did not generate a result file.
2025-08-11T04:00:39.5327102Z   >>>
2025-08-11T04:00:39.5327237Z 
2025-08-11T04:00:39.5327430Z Summary: 5862 tests, 1 error, 1 failure, 1216 skipped
```

**To Reproduce**
https://github.com/ros-controls/ros2_control_ci/issues/445

christophfroehlich, 2025-08-31:
https://github.com/ros-controls/ros2_control_ci/issues/453

christophfroehlich, 2025-09-02:
https://github.com/ros-controls/ros2_control_demos/issues/898

christophfroehlich, 2025-09-28:
https://github.com/ros-controls/ros2_control_ci/issues/482

christophfroehlich, 2025-09-30:
https://github.com/ros-controls/ros2_control_ci/issues/485

jsantoso91, 2025-10-01:
Hi @christophfroehlich, do you think updating the CMakeLists.txt to use add_ros_isolated_launch_test like shown in this page would help addressing the issue?
https://docs.ros.org/en/rolling/Tutorials/Intermediate/Testing/Integration.html

christophfroehlich, 2025-10-01:
maybe yes. we have it with example_13 already, and this one did not fail in any of the linked issues above. 
Do you want to give it a try?