ros-controls/ros2_control_demos #886: Difference between control methods in example 1 and 6

tim-yitian-xian, 2025-08-21:
Hi all, I used ROS1 before and I feel quite familiar with the control methods from here: https://github.com/ros-controls/ros2_controllers/blob/master/ros2_controllers_test_nodes/ros2_controllers_test_nodes/publisher_joint_trajectory_controller.py, which basically implements a controller that publishes the joint trajectory to the corresponding topic.

Therefore, I am wondering what are the benefits of controller implemented in example 6? I guess it is for real-time or industial standard? And how can I implement the same function in example 1 but in the format of example 6?

Many thanks for your help!

christophfroehlich, 2025-08-21:
in ex1, all actuators are implemented in the same hardware component while in ex6, there is a separate hardware component plugin for every actuator.
Apart from that, ex1 additionally has configured a joint_trajectory_controller. There is no new controller in ex6, so what do you mean?

tim-yitian-xian, 2025-08-25:
@christophfroehlich thanks for your reply! Could you please briefly talk about their differences, the way implemented in ex1 and ex7?

christophfroehlich, 2025-08-26:
I already did. Example 1 uses joint_trajectory_controller, while example 7 showcases how to write a custom controller.